IHP-GmbH/IHP-Open-PDK #352: Compile for Xyce models fails

hpretl, 2025-02-10:
Since the merge of 9681b01220f06998d2f43170f8187d4f2b2fb501 using the script `adms-compile-va.sh` fails with an error:
```
buildxyceplugin is building a plugin from the Verilog-A files:
psp103.va
The plugin name will be generated automatically from the module name
in the last file of this list.
Building C++ for Verilog-A input file 'psp103.va'...
C++ for Verilog-A input file 'psp103.va' failed
See buildxyceplugin.log for details
```

When I delete the file `psp103/discipline.h` then everything works fine.

@dwarning Any idea what might the issue? I have tried with Xyce 7.8 and 7.9, no difference.

hpretl, 2025-02-10:
I should add: When removing the local `discipline.h` in `psp103` and `r3_cmc` everything is fine.

dwarning, 2025-02-10:
@hpretl 
Can you attach one of the failed buildxyceplugin.log.

For me it works. 
I checked the both discipline.h - they are OK. If I remove the header files it compiles for me too because both compiler are using their own copy. So they are obsolete.

I have my own Xyce and admsXml build here on my Ubuntu machine:
`dietmar@modsys:~/Projects/IHP-Open-PDK/ihp-sg13g2/libs.tech/verilog-a$ Xyce -v
Xyce DEVELOPMENT-202412082152-(Public_Release-7.8.0-128-g5b74b0d5-dirty)-opensource
dietmar@modsys:~/Projects/IHP-Open-PDK/ihp-sg13g2/libs.tech/verilog-a$ admsXml -v
[usage..] <release name="admsXml" version="2.3.7" date="Nov 10 2024" time="11:22:18"/>
`
Do you changed something with admsXml installation?

hpretl, 2025-02-10:
```
/tmp > xyce -v
Xyce DEVELOPMENT-202501121412-(Public_Release-7.8.0-326-gbe78559c)-opensource
/tmp > admsXml -v
[usage..] <release name="admsXml" version="2.3.7" date="Apr 22 2024" time="14:42:32"/>
```
No change with admsXml or otherwise, just a plain build of Xyce.

Here is the `buildxyceplugin.log`:
```
/tmp/IHP-Open-PDK/ihp-sg13g2/libs.tech/verilog-a/psp103 > cat buildxyceplugin.log
buildxyceplugin log begun at Mon Feb 10 07:39:02 PM CET 2025

 script invoked as:
 /foss/tools/bin/buildxyceplugin psp103.va ../../xyce/plugins
 ----------------
Building C++ for Verilog-A input file 'psp103.va'...
[info...] -x: skipping any implicit xml scripts
[info...] admsXml-2.3.7 (unknown) Apr 22 2024 14:42:32
[fatal..] discipline.h: during lexical analysis syntax error at line 21 -- see ';'
```

I compared with the existing `discipline.h` and the error is caused by the `;` at the end of the lines `discipline xxx;` and `nature xxx;`

hpretl, 2025-02-10:
@dwarning Compared with this here https://verilogams.com/refman/basics/disciplines.html there should indeed be no `;` at the end of the lines.

hpretl, 2025-02-10:
Here is my fix: `sed -E '/^(nature|discipline)/s/;( *$)//'`

dwarning, 2025-02-10:
@hpretl 
Most developments for Verilog-A references to [Language Reference Manual](https://www.accellera.org/images/downloads/standards/v-ams/VAMS-LRM-2-4.pdf).
On page 387 we see the ";" on the end.
I am using these files for years in ngspice/adms, Xyce/adms and OpenVAF. It is first time I see this problem - opposite to the backslash before logic keyword on same page 387 which I removed long time ago.

hpretl, 2025-02-10:
Ah, now I have an idea: which version of `adms` are you using? I use the one which can be installed via APT:
```
/foss/designs > apt search adms
Sorting... Done
Full Text Search... Done
adms/noble,now 2.3.7-1build1 arm64 [installed]
  Automatic device model synthesizer for Verilog-AMS

libadms0/noble,now 2.3.7-1build1 arm64 [installed,automatic]
  Shared library for automatic device model synthesizer
```

KrzysztofHerman, 2025-02-10:
@hpretl @dwarning in order to make this thread even more interesting I can tell you that I have successfully compiled the models using  `adms` and `openvaf`. However `adms` deletes the `discipline.h` files in both directories.

hpretl, 2025-02-10:
... oh my... :-)

hpretl, 2025-02-10:
@dwarning I have a feeling that both ADMS (and OpenVAF) are in serious need of an active maintainer if we build on it:

From the Xyce homepage:
[image]

Looing at this referenced project... no update since 3 years.

[image]

hpretl, 2025-02-10:
As said, I am really no expert here, but: Can't we retire ADMS for good and just keep using and improving OpenVAF?

dwarning, 2025-02-11:
@hpretl 
It is true ADMS is not updated since years. But your apt install is more outdated because it has not the commit 13fdce from Neal Wood from 04.01.22 "Fix bad grammar rule for optional semicolon".
In case you want use Xyce/adms I would prefer to build your own admsXml from github repository.

hpretl, 2025-03-10:
Update: With a compile from sources ADMS works properly, so we can close this one.

closed 2025-03-10 by hpretl