IHP-GmbH/IHP-Open-PDK #356: ngspice Too few parameters for subcircuit type "npn13g2l"

jalcim, 2025-02-12:
when i follow this tutorial : https://ihp-open-pdk-docs.readthedocs.io/en/latest/analog/ngspice.html#ngspice-installation

i have this error, i have try to change my parameter, but its change nothing.

```
Note: No compatibility mode selected!

Circuit: **.subckt dc_hbt_13g2

Too few parameters for subcircuit type "npn13g2l" (instance: xxq1)
    Simulation interrupted due to error!

Note: No ".plot", ".print", or ".fourier" lines; no simulations run
```

i have try
```XQ1 net2 net1 GND GND npn13G2l Nx=1 le=2.50e-6 dtemp=0 Ny=1 we=0.12e-6 El=2.5 selft=1```
and
```XQ1 net2 net1 GND GND npn13G2l Nx=1 le=2.50e-6 dtemp=0 Ny=1 we=0.12e-6 selft=1```
and a lot of other... no way.

documentation require update ?
or its a misstake of my part ?

KrzysztofHerman, 2025-02-12:
@jalcim could you please share your whole netlist ant the ngspice and the  PDK version (branch, commit) ? 
I have just executed the sample netlist from the documentation and it works. 
```
.subckt npn13G2 c b e bn 
.param Nx=1 dtemp=0
+Ny=1 le=0.96e-6 we=0.12e-6
+El=le*1e6
+selft=0
```
`El` and `le` can be used interchangeably keeping in mind the units `um` and `m` respectively.  The `xschem` and `qucs-s` symbols expose only `Nx `and `El` to the designer.

jalcim, 2025-02-12:
i do all like the documentation.

but i gonna resume each step.

```
sudo apt-get install -y build-essential
sudo apt-get install -y qtbase5-dev qttools5-dev
sudo apt-get install -y clang cmake libtool autoconf
sudo apt-get install -y python3 python3-dev python3-pip python3-virtualenv python3-venv
sudo apt-get install -y ruby ruby-dev
sudo apt-get install -y btop tree xterm graphviz git
sudo apt-get install -y octave liboctave-dev
sudo apt-get install -y python3-sphinx python3-sphinx-autoapi python3-pandas python3-tk python3-pytest
sudo apt-get install -y libqt5xmlpatterns5-dev qtmultimedia5-dev libqt5multimediawidgets5 libqt5svg5-dev libqt5opengl5
sudo apt-get install -y tcl8.6 tcl-dev tcl8.6-dev
sudo apt-get install -y tk8.6 tk8.6-dev
sudo apt-get install -y flex clang gawk xdot pkg-config bison curl help2man perl time
sudo apt-get install -y libxpm4 libxpm-dev libgtk-3-dev libffi-dev
sudo apt-get install -y libjpeg-dev libfl-dev libfl2
sudo apt-get install -y libreadline-dev gettext
sudo apt-get install -y libboost-system-dev libboost-python-dev libboost-filesystem-dev zlib1g-dev
sudo apt-get install -y libx11-6 libx11-dev
sudo apt-get install -y libxrender1 libxrender-dev
sudo apt-get install -y libxcb1 libx11-xcb-dev
sudo apt-get install -y libcairo2 libcairo2-dev libxaw7-dev
sudo apt-get install -y libgz libfl2 libfl-dev zlibc zzlib1g zlib1g-dev libz-dev libgit2-dev
sudo apt-get install -y libgoogle-perftools-dev
sudo apt-get install -y gengetopt groff pod2pdf libhpdf-dev
sudo apt-get install -y libfftw3-dev
sudo apt-get install -y libxml-libxml-perl libgd-perl
sudo apt-get install -y libsuitesparse-dev gfortran swig libspdlog-dev libeigen3-dev liblemon-dev
git clone --recursive https://github.com/IHP-GmbH/IHP-Open-PDK.git
echo "export PDK_ROOT=\$HOME/.tools/IHP-Open-PDK" >> ~/.bashrc
echo "export PDK=ihp-sg13g2" >> ~/.bashrc
echo "export KLAYOUT_PATH=\"\$HOME/.klayout:\$PDK_ROOT/\$PDK/libs.tech/klayout\"" >> ~/.bashrc
echo "export KLAYOUT_HOME=\$HOME/.klayout" >> ~/.bashrc
source ~/.bashrc
```

```
git clone https://git.code.sf.net/p/ngspice/ngspice ngspice-ngspice
cd ngspice-ngspice
./autogen.sh
./configure --enable-osdi
make
sudo make install
ngspice -v
```

version 44.2

```
emacs ihp-sg13g2/libs.tech/ngspice/models/dc_hbt_13g2.spice
```

```
**.subckt dc_hbt_13g2
Vce net3 GND 1.2
I0 GND net1 1u
Vc net3 net2 0
.save i(vc)
XQ1 net2 net1 GND GND npn13G2l Nx=1 le=1.0e-6

.lib cornerHBT.lib hbt_typ

.param temp=27
.control
save all
op
print I(Vc)
.endc

.GLOBAL GND
.end
```

output

```
Note: No compatibility mode selected!

Circuit: **.subckt dc_hbt_13g2

Too few parameters for subcircuit type "npn13g2l" (instance: xxq1)
    Simulation interrupted due to error!

Note: No ".plot", ".print", or ".fourier" lines; no simulations run
```

KrzysztofHerman, 2025-02-12:
@jalcim thank you for the detailed response, What about the PDK version ?

sergeiandreyev, 2025-02-12:
from the log above it should be the latest `main` branch of the OpenPDK

jalcim, 2025-02-12:
> from the log above it should be the latest `main` branch of the OpenPDK

its the latest 'main' branch of the OpenPDK

jalcim, 2025-02-12:
Oh...

Alright, it's working now! I hadn't naturally considered that the development branch might be the more stable one.
May I suggest creating a new tag? It could help ensure that newcomers following the PDK don’t run into frustration.

Thanks for help.

closed 2025-02-12 by jalcim