IHP-GmbH/IHP-Open-PDK #360: NGSPICE basic testcases : Unknown model type pspnqs103va

lionelsaintecluque, 2025-02-13:
Hello, 

In [4. Simulation using NGSPICE](https://ihp-open-pdk-docs.readthedocs.io/en/main/analog/ngspice.html#simulation-using-ngspice)

**Yesterday**, first example (BJT testcase) was failing in main branch. I switched to dev branche and got successful results. (according to issue 356 : https://github.com/IHP-GmbH/IHP-Open-PDK/issues/356)
(Thanks for your fast answer)

**Today** I tryied the second example (mosfet) and got another failure. 

I'm runnin dev branch of [IHP-Open-PDK](https://github.com/IHP-GmbH/IHP-Open-PDK)
I tested with ngspice ngspice-36 and ngspice-44.2 (similar failure). 

I copyied and pasted the proposed example in a file (toto.sp), placed in the same directory as the libraries (it's a bit dirty, but I wanted to test fast).
I ran the simulation within that directory .

```
... models % pwd
... /IHP-Open-PDK/ihp-sg13g2/libs.tech/ngspice/models

... models % cat toto.sp 
*Test 2

.lib cornerMOSlv.lib mos_tt
Vgs net1 GND 0.4
Vds net3 GND 1.0
Vd net3 net2 0
.param temp=27
XM1 net2 net1 GND GND sg13_lv_nmos w=1.0u l=0.13u ng=1 m=1
.control
save all
op
let Id = @n.xm1.nsg13_lv_nmos[ids]
print Id
.endc
.GLOBAL GND
.end

... models % ngspice  -b toto.sp 

No compatibility mode selected!

Circuit: *lib cornerMOSlv.lib mos_tt

Warning: unknown device type: n
Warning: unknown device type: n
Warning: unknown device type: n
Warning: Model issue on line 0 :
  .model xm1:sg13g2_lv_nmos_psp pspnqs103va type=+1 level=103.60 tr=27.0 d ...
Unknown model type pspnqs103va - ignored
Warning: Model issue on line 0 :
  .model xm1:sg13g2_lv_pmos_psp pspnqs103va type=-1 level=103.60 tr=27.0 d ...
Unknown model type pspnqs103va - ignored
Error on line 0 :
  n.xm1.nsg13_lv_nmos net2 net1 s b sg13g2_lv_nmos_psp w=    1.000000000000000e-06     l=    1.300000000000000e-07     nf=    1.000000000000000e+00     mult=    1.000000000000000e+00     as=    3.400000000000000e-13     ad=    3.400000000000000e-13     ps=    2.680000000000000e-06     pd=    2.680000000000000e-06     dta=    0.000000000000000e+00     ngcon=2
 unknown device type - error 
    Simulation interrupted due to error!

Note: No ".plot", ".print", or ".fourier" lines; no simulations run

```

Did I do something wrong ? Did I missed a step ? 

Thanks by advance

KrzysztofHerman, 2025-02-13:
@lionelsaintecluque 
ngspice from the version 41 supports the osdi interface what enables it to simulate model compiled from Verilog-A code. Our  `mosfets` and `polysilicion resistors` use Verilog-A models which have to be compiled using the compiler `openvaf`  (ngspice) and/or `adms ` (Xyce). The easiest way is to install openvaf and execute the script `openvaf-compile-va.sh` from `libs.tech/verilog-a`. Additionally you have to create a symbolic link `.spiceinit` in your `/home/$USER/` directory which points to the `$PDK_ROOT/$PD/libs.tech/ngspice/.spiceinit`.

lionelsaintecluque, 2025-02-13:
OK (I'm very old school, Xspice looked so new to me that I never really used it). 

Then, I will be interested to know if anyone using macosx have advice to give to me to install such tools. I naively tried to install ADMS from https://github.com/Qucs/ADMS, but it does not really compiles with CMAKE (problems with yacc ? )

I'm affraid I may face such difficulties with other verilog-A compilers (I am not sure open-vaf compiles well on macos-X ?) 

If someone have feedbacks to share it save me hours or days I feel.

KrzysztofHerman, 2025-02-13:
@lionelsaintecluque 
I have to say that the open source EDA tools are pretty sensitive to the platform and that is why almost everyone is using the latest ubuntu LTS version. Even if you take a look at the proprietary tool you'll see that it is similar story. You can ask for some more support on this [slack channel](https://join.slack.com/t/open-source-silicon/shared_invite/zt-2zsrkvm4l-T9QLgyHfMwl4TjPodfp3xg)

jalcim, 2025-02-13:
I can't find any trace of this in the documentation.

https://ihp-open-pdk-docs.readthedocs.io/en/latest/analog/ngspice.html#ngspice-basic-testcases

KrzysztofHerman, 2025-02-13:
@lionelsaintecluque you are right however if you use Qucs-S or xschem the `install.py` scripts do it under the hood. 
https://ihp-open-pdk-docs.readthedocs.io/en/latest/analog/xschem.html

jalcim, 2025-02-13:
i prepare a pull request for changing documentation.
i add that and other things inside.

lionelsaintecluque, 2025-02-14:
@KrzysztofHerman : Somehow I find proprietary tools easier to install, because there is somehow no discussion about platform. For example, if a windows computer is required, the question is just who pays... 

Regarding macosX and EDA software, it's often a bit tricky. There might be several installing options : homebrew, macports, or compile by myself.
The last one is now my preferred option, but care must be taken for dependencies : if homebrew or macport provides those dependencies, ./configure (or similar) may try to use them, making it all entangled.

The drawback of compiling by myself, is the time. That's why I was curious of people feedbacks.

To be specific, regarding open-vaf, there is apparently a dockers container availaible, but it fails to compile. I may insist in that direction before getting resigned ans find a linux server somewhere.

jalcim, 2025-02-14:
The reality is, its so complex ton install when you dont know exactly what to do.
In fact we need build by source a loot of tools (like llvm for openVAF).
This technical dependency is verry big. 
I work actually at a full script for offering a more easy solution, but i cannot make it compatible with all operating system.

KrzysztofHerman, 2025-02-14:
@lionelsaintecluque a lot of substantial progress was done during the recent few years in the open source silicon domain and the reality is that behind each open EDA tool PDK there are individuals or small groups of people working or it. We are not an exception.  The good thing is that this is open source so anybody can pick up the solution and make it work on any platform. We simply do not have the resources to develop and test it using other platforms. 

@jalcim thank you for the contribution to our documentation. I will review it. We are also happy to see the solution, which simplifies the build of the tools, which is not an easy task, especially for newcomers.

holvo, 2025-02-14:
OpenVAF has never been compiled under macOS. You will find its most recent source code, pre-compiled executables  and some compile instructions for Linux and Windows at https://github.com/OpenVAF/OpenVAF-Reloaded .

jalcim, 2025-02-14:
Could you please clarify which tag would be the most relevant for compiling OpenVAF in the context of IHP?

adrienluitot, 2025-02-19:
> OpenVAF has never been compiled under macOS. You will find its most recent source code, pre-compiled executables and some compile instructions for Linux and Windows at https://github.com/OpenVAF/OpenVAF-Reloaded .

@lionelsaintecluque @holvo I actually was able to compile OpenVAF on MacOS. The link you shared here, is compatible with MacOS. As far as I remember I pushed the fixes.
And the compilation procedure is pretty much the same as on linux. It is also possible do like the [MacOS CI](https://github.com/OpenVAF/OpenVAF-Reloaded/blob/master/.github/workflows/rust-ci.yml) (part `macos-build-test`).
With this OpenVAF version you will need to use ngspice 44 (or more recent), because this version only generates OSDI v0.4.

___________

> Could you please clarify which tag would be the most relevant for compiling OpenVAF in the context of IHP?

@jalcim The most basic compilation should work, you don't need to specify special arguments/tags.
However, you must be careful with the versions of the different tools.

jalcim, 2025-02-20:
> , is compatible with MacOS. As far as I remember I pushed the fixes.

i think when we talk about MacOS its can be relevant to specify if its intel or arm ship.

>, The most basic compilation should work, you don't need to specify special arguments/tags.
However, you must be careful with the versions of the different tools.

for me its didn't work, easily, hopefully im a very experimented linux user, so i have rebuild all the tools required.

Actually its didnt work totaly for me, but i have try 1 month before.
I need to retry, and make a detailled issue, with a pull request for changing documentation i think.

I have a lot of problem with OpenVAF especialy, exept some tag, he require a so new llvm version for my debian by example.

lionelsaintecluque, 2025-03-03:
Hello all, 

@adrienluitot : I did not installed openvaf-reloaded on my mac. It requires to install llvm, and I don't have the courage by now... But thanks for your work. 

I installed openvaf on my fedora station, and I simulate in this computer for analog things. 

Thanks for your help all, it's working now !

sergeiandreyev, 2025-03-10:
Hi @lionelsaintecluque, is there anything else that should be fixed as a part of this issue, or we can close it?

adrienluitot, 2025-03-13:
> did not installed openvaf-reloaded on my mac. It requires to install llvm, and I don't have the courage by now... But thanks for your work.

I think installing it with brew is easy? I don't really recall 100% but I have LLVM 19 installed with brew, so I guess you could do the same?

sergeiandreyev, 2025-05-21:
> Hi [@lionelsaintecluque](https://github.com/lionelsaintecluque), is there anything else that should be fixed as a part of this issue, or we can close it?

@lionelsaintecluque, gentle reminder..

jalcim, 2025-05-22:
Hello, sorry we forget because we have use a verry tricky hack, so no, its not really solved.

sergeiandreyev, 2025-05-23:
@dwarning, could you please provide the link to the latest OpenVAF version repository which is supported by IHP OpenPDK? as I remember it was Arpad Bruemen fork..

thonglinh90, 2025-06-29:
I have the same issue. Should version of OpenVAF I used?

jalcim, 2025-07-28:
> See the link to binaries on this fork [OpenVAF](https://github.com/arpadbuermen/OpenVAF).

no its this one https://github.com/OpenVAF/OpenVAF-Reloaded