IHP-GmbH/IHP-Open-PDK #362: Vagueness in layer relations in layout_rules.pdf

MilosMitrovic411, 2025-02-16:
I have a few questions on how you define relations between shapes and layers.

- unrelated - two regions which do not touch each other
&
- abut - two edges of two different layers touching each other

What does "touch" mean? Since you defined abut as you did, does "touching" represent any two point of a two different shapes interacting. If so, then "unrelated" shapes would be the ones who do not "interact" at all with one another, right?

- nBuLay - (((NWell ≥ 3.0 µm) sized by 1.0 µm/side) OR nBuLay:drawing) AND NOT nBuLay:block

Does "sized by 1.0 µm/side" represent "sized down" or  "sized up"?

- inside & not inside

What is your definition of those? In the picture bellow, N+Activ and ThickGateOxide are "overlapping", would you say that N+Activ is "inside" or "not inside" of ThickGateOxide, are they "touching"? For an example, should rule NW.d or NW.d1 be applied to this N+Activ, or should rule NW.d apply to (N+Activ AND NOT ThickGateOxide), and rule NW.d1 apply to (N+Activ AND ThickGateOxide)?

[image]

stefansimon42, 2025-02-21:
Hi @MilosMitrovic411 ,

The definition of layer relationships is a difficult topic, as personal preference often plays a role and the definitions of different tool vendors can differ widely. For this reason, we have included a chapter in the Layout Rule documentation that explains the important checks. Of course, this overview is not yet complete and can be expanded at any time.

Regarding your questions:

1. Perhaps this description should be updated in the future.   The word touch can be a bit confusing here. Certain design tools use the keyword “touch” when two edges have a distance of exactly 0.
_unrelated_: Two regions are unrelated if they do not touch or overlap in any way at any time. 
_abut_: Two regions touch or overlap each other (see abutted ties).
The tool Calibre uses the term “INTERACT” for this. I think that describes it best.

2. You're right. It is “sized down”, i.e. “sized by -1.0”. I will fix that.

3. With regard to your picture, I would say: N+Activ is “not inside” PSD. Inside means that layer A is completely enclosed by layer B. In contrast, outside means that layer A is unrelated (see above) to layer B.

I hope my answers will help you.

MilosMitrovic411, 2025-02-21:
Hi @stefansimon42,

Thank you very much for your in depth explanation.

It is of a great help. It basically aligns with how I saw it, but as you said it is somewhat personal preference, so I wanted to be sure.

Just one more question about the nBuLay, since it is sized down from NWell, is the DRC error on the picture bellow correct?
[image]

Should I even check nBuLay rules when the layer itself is not explicitly drawn?

Thank you again @stefansimon42 for your time.

stefansimon42, 2025-02-24:
From my personal perspective, you should also check automatically generated nBuLay. In addition, a “dummy violation” could be added that displays all generated nBuLay.

Unfortunately, I cannot reliably judge the correctness of rule implementations based on screenshots. It would be much more helpful if you could provide a GDS file.

MilosMitrovic411, 2025-02-24:
Thank you. Here you go an example GDS file [ex.zip](https://github.com/user-attachments/files/18947215/ex.zip).

My main concern with this example is that even though N+Activ and nBuLay drawing layers are unrelated, they are electrically connected through the NWell, so my question is basically should the rule NBL.e, as well as NBL.f, be tested inside NWell or not?