IHP-GmbH/IHP-Open-PDK #376: MIM.i rule in magic tech

smunaut, 2025-03-05:
First, that rule isn't super clear in the layout guide. It's in the diagram but there is no accompanying text.
Second, it seems the rule triggers when under `mimcap` but not `mimcapcontact` which is probably an omission.
Finally should that rule also trigger for any contacted `metal5` bottom plate if the reason for the rule is antenna related ?

(cc @RTimothyEdwards )

sergeiandreyev, 2025-03-05:
please double check - the `MIM.i` was removed in the DRM v0.2:

[image]

FYI, @stefansimon42

RTimothyEdwards, 2025-03-05:
The rule for `via_under_mimcap via4` should have `and *mimcap` instead of `and mimcap` so that it includes areas of MiM cap contacts.

However, if the rule has been deprecated, then the rule and associated cif layer should just be removed.

smunaut, 2025-03-05:
@sergeiandreyev Ok, but then Figure 6.13 should also be edited to remove the corresponding drawing.

sergeiandreyev, 2025-03-05:
@smunaut, agree

stefansimon42, 2025-04-01:
Hi @smunaut,
The figure will be updated in the next DRM revision.