IHP-GmbH/IHP-Open-PDK #385: IOPadIOVss and IOPadVss don't have strong `TopMetal2` connections.

smunaut, 2025-03-07:
If you look at the Vdd pads, they have very strong connection to the rings through TopMetal2.

For instance here's `IOPadIOVdd`, you can see the direct connection on the bottom pad.
[image]

But the top pad is a `IOPadIOVss` which should connect to the `IOVss` ring which is right next to it, but instead of having a strong connection through the thick `TopMetal2`, it goes all the way down to `Metal2`, through quite the via stack ...

What I think it should be doing is something like: 

[image]

(this is just an edited image, I haven't implemented it).

Same issue is present for `IOPadVss`. The `IOPadVdd` has a strong connection cutting the through IO ring on `TopMetal2` to directly reach the `Vdd` ring, but the `Vss` one does not and again goes through a lot of via and down to a thin lower metal layer.

sergeiandreyev, 2025-03-10:
Hi @stafverhaegen-chipflow, FYI

stafverhaegen-chipflow, 2025-03-11:
I have a quick fix in my code for `IOPadIOVss`. I can provide gds but this is without the ptap changes that have been done at IHP.
For `IOPadVss` I am holding off for more analysis and the change is a little more involved. I did use top metal for `IOPadVdd`  as there Metal2 was used for internal routing.
Also weak is I think an exaggeration, I would call is non-ideal :) `IOPadVss` is meant for digital core ground and I expect IR-drop problems on the logic before electromigration problem due to current density. Said otherwise if the number of `IOPadVss`/`IOPadVdd` pairs would be so low that one would have electromigration problems due to the high current I would also expect IR-drop problems in the core.

sergeiandreyev, 2025-03-17:
> I can provide gds but this is without the ptap changes that have been done at IHP
we should avoid mixing two different approaches..

smunaut, 2026-10-22:
I saw IOPadIOVss was fixed, tbh that's fine for me.

However there is the same issue in CMOS5L now with TM1 and that is not fixed.

b10346, 2026-10-26:
@smunaut I will take a look at CMOS5L IOs. I would also prefer to have separate issues (requests) regarding SG13G2 and CMOS5L.

b10346, 2026-11-04:
@smunaut PR is opened

Mauricio-xx, 2026-11-10:
IOPadIOVss was fixed for sg13g2 earlier and is now fixed for sg13cmos5l by #1330, merged into dev. IOPadVss stays as discussed above. Please reopen or open a new issue if something is still missing.

closed 2026-11-10 by Mauricio-xx