IHP-GmbH/IHP-Open-PDK #389: Feature request: Add nmos3 / pmos3 symbols

smunaut, 2025-03-08:
Those variant were in sky130 where the bulk connection is set by a property rather than a wire/port. It was _so_ convenient to keep schematics clean, I'd love to see them for SG13G2 too.

KrzysztofHerman, 2025-03-10:
@smunaut thank you for this observation. Indeed we could provide this having two options GND/sub! for the bulk pin. Right now we are focused on bug fixing so we will take some time to do so. If you are willing to provide such symbols we will be happy to accept your PR.

hpretl, 2025-03-10:
@KrzysztofHerman If this is implemented (and I think this would be good to have) then please make the bulk node user-specific, like adding an additional property like $W$ and $L$ (for sure Xschem has a way to nicely set this @stefanschippers). As a circuit designer I do not want to be forced to name a VDD/VSS net for bulks a specific way.

KrzysztofHerman, 2025-03-11:
@hpretl 
We could provide something like this:
```
lvs_format="@spiceprefix@name @pinlist @body @model l=@l w=@w ng=@ng m=@m"
format="@spiceprefix@name @pinlist @body @model w=@w l=@l ng=@ng m=@m"
template="name=M1
l=0.45u
w=1.0u
ng=1
m=1
body=GND
model=sg13_lv_nmos
spiceprefix=X
"
```
This way you could specify either is a GND, VSS, VDD, sub! or any other net you wish to use.

smunaut, 2025-03-11:
Yeah, that's exactly what I had in mind.
I was waiting for the nmos direction pr to be merged and then try and make the various nmos3/pmos3 symbols following that scheme.
(I'll just need to move some of the annotations to make room to show the body net)

StefanSchippers, 2025-03-11:
These devices are done in sky130 with the names of nfet3_*.sym and pfet3_*.sym.

The body terminal is specified ( as written by @hpretl ) with a body attribute (like body=GND or body=VCCA).
Within the symbol add `@body` after @pinlist (@pinlist only shows drain, gate, source terminals)
`format="@spiceprefix@name @pinlist @body sky130_fd_pr__@model L=@L W=@W`

[image]

hpretl, 2025-03-11:
Yes, using a property and showing the value on the symbol is precisely how I think it should be.

smunaut, 2025-03-28:
I drew those :

[image]

I can open a PR if they're suitable.