IHP-GmbH/IHP-Open-PDK #390: xschem NMOS symbol has the source up by default

smunaut, 2025-03-08:
Isn't it a bit weird the NMOS symbol default orientation has the source on top ?

hpretl, 2025-03-10:
@stefanschippers Can you please look at the proposed implementation? Is there a way to implement this w/o screwing up existing schematics?

KrzysztofHerman, 2025-03-11:
@hpretl, @StefanSchippers just to confirm.  

The existing schematics should have the mosfets instantiated using the following entry in *.sch file:
```
C {sg13g2_pr/sg13_lv_nmos.sym} 360 -250 2 1 {name=M1
```
After applying 180 rotation and one horizontal flip it changes to:
```
C {sg13g2_pr/sg13_lv_nmos.sym} 360 -250 0 0 {name=M1
```
so maybe this would be the solution to rework all the schematics in a bulk ?

hpretl, 2025-03-11:
@KrzysztofHerman This is precisely how I would do it, either with a script or a text editor, definitely no mousing involved :-)

KrzysztofHerman, 2025-03-11:
@hpretl what about the default orientation of `pmos`, should be same as `nmos` or 180 rotated ?

StefanSchippers, 2025-03-11:
> [@hpretl](https://github.com/hpretl), [@StefanSchippers](https://github.com/StefanSchippers) just to confirm.
> 

Yes , see the effect  {rot flip} = {2 1} (**right**) vs {0 0} (**left**)

[image]

smunaut, 2025-03-11:
IMHO pfets should have their source on top like they do now when in the default orientation.

hpretl, 2025-03-11:
@KrzysztofHerman Default configurations should be (IMHO):

NMOS: drain up, source down, gate left, bulk right
PMOS: source up, drain down, gate left, bulk right

Think of drawing an inverter, NMOS/PMOS should come in the correct orientation by default.

StefanSchippers, 2025-03-11:
> NMOS: drain up, source down, gate left, bulk right
> PMOS: source up, drain down, gate left, bulk right

Yes, this is the default in all cmos processes I have seen, including sky130.
GF180MCU also has this "reversed" nmos issue.

KrzysztofHerman, 2025-03-11:
This is the proposal for the updated symbols. 
It modifies the default values of `w` and `l` and set it to the minimum. 

[image]

Is there any more observation ?

StefanSchippers, 2025-03-11:
just a tiny aesthetic observation. if possible move a veeeery little bit down (by reducing the grid size with 'g' key and moving down) the m=... and rfmode=... texts , so you can add a wire to the 'b' terminal without running over the texts. I think it is possible since there is a little room above the model name text.

hpretl, 2025-03-11:
@StefanSchippers @KrzysztofHerman Yes, that would be super nice: Move the texts on all symbols a little bit so that a wire running out of bulk does not go through a text.

hpretl, 2025-03-11:
@KrzysztofHerman the "S" sometimes seems off: 

[image]
 compared to 

[image]

hpretl, 2025-03-11:
BTW, excuse my ignorance: What is `rfmode=1` and why is `Lmin` then 0.72u?

StefanSchippers, 2025-03-11:
@KrzysztofHerman to reposition the S and D labels descend into a 'good' symbol, select both the S and D texts and do a ctrl-c to copy. Then go into a symbol with the wrong placements, delete both texts and do a ctrl-v to paste. SOom or pan a little bit and you should see the position where copied objects need to go. see short video. 

https://github.com/user-attachments/assets/0b28ea1e-143a-42a6-8cce-5a97b3774262

smunaut, 2025-03-11:
Look mostly good to me.
I'm possibly bike-shedding, so feel free to ignore this, but my 2ct:
- I like the little 'ball' for the PMOS. Yes the arrow differentiates already but I have an easier time spotting the ball when a PMOS is used in more complex circuit than an inverter.
- Maybe a bit thinker line to mark the gate oxide of HV devices ?

Again, I understand this is purely personal preference, it's just what I noticed I preferred when I compared to the sky130 symbols and I figured it might be good to point them out if modifying the symbols.

KrzysztofHerman, 2025-03-12:
> BTW, excuse my ignorance: What is `rfmode=1` and why is `Lmin` then 0.72u?

So the symbol with `rfmode=1` annotates it in the netlist as follows: 
```
XM6 net17 net18 net19 net20 sg13_lv_pmos W=1.0u L=0.72u ng=1 m=1 rfmode=1
XM8 net21 net22 net23 net24 sg13_hv_pmos W=1.0u L=0.72u ng=1 m=1 rfmode=1
```
The effect you got is that the simulation will be performed using some more parameters 
extracted using the layout of rfmos.

Here a piece of a model card. 

[image]

KrzysztofHerman, 2025-03-12:
Here goes updated version of the symbols:

[image]

Changes:

- position of `D S B` pin labels is now unified
- position of the text is aligned such way that one can wire the bulk without overlapping the text layer 
- for nmos devices I have flipped horizontally the bulk triangle and shifted the arrow in the source to the maximum right position.  

Thank you @StefanSchippers for the video, it was very helpful. 

There is also a proposal to make this versions of symbols `sg13_lv_nmos4` and `sg13_lv_nmos3`

[image]

For 3-terminal device the `body` terminal will be exposed to the user as an instance parameter
```
K {type=nmos
lvs_format="@name @pinlist @body @model w=@w l=@l ng=@ng m=@m"
format="@spiceprefix@name @pinlist @body @model w=@w l=@l ng=@ng m=@m"
template="name=M1
l=0.13u
w=0.15u
ng=1
m=1
body=GND
model=sg13_lv_nmos
spiceprefix=X

```

hpretl, 2025-03-12:
@KrzysztofHerman Thanks, looks good to me :-)

KrzysztofHerman, 2025-03-12:
So maybe this version would be ok? I have diminished the size of the label and also put it into center.

[image]

smunaut, 2025-03-12:
I don't mind the font size too much the the center is annoying.

For instance the ability to pass a wire through the gate like when you draw a current mirror with multiple outputs is very convenient (and one of the reason I like the mos3 variants of sky130) :

[image]

StefanSchippers, 2025-03-12:
In xschem the **!** character does not have any special meaning (If i remember well Cadence uses that for global nodes).

In xschem nodes become global if they are attached to tier symbols (vdd.sym or gnd.sym) or label symbols with instance attribute `global=1` defined.

That said it is perfectly fine to use **sub!** for the body terminal. Just use **sub!** everywhere in the symbol attributes.

KrzysztofHerman, 2025-03-12:
@smunaut good point, I will revert the changes. 
@StefanSchippers we use `sub!` as a global net in the proprietary PDK. I have also provided a symbol of it and it annotates GLOBAL sub! net in the netlist. 

There is one more proposal to use the following symbol convention:

[image]
 
Here `nmos` is a device name consistent with the device names in our PyCells where we have:
`nmos, nmosHV, pmos, pmosHV, rfnmos, rfpmos, rfnmosHV, rfpmosHV`. 
of course the spice and cdl netlist will use a valid model name to annotate the device. 

We could name the symbols using the device naming scheme ie. `nmos.sym`, `nmos4.sym`, `nmos3.sym`

hpretl, 2025-03-12:
@KrzysztofHerman Cleaning up device name is certainly an option.

smunaut, 2025-03-12:
Watchout that in the proposal you have above, you have `LVT` marker which I wouldn't consider appropriate given each transistor type only have one Vt variant.

sergeiandreyev, 2025-03-12:
I guess, in that example was meant Low Voltage (we have LV and HV MOSFETS), not the threshold type like in Sky130

smunaut, 2025-03-12:
The oxide thickness bar already represents that.
Adding a `LV` or `HV` marker would be fine if you want to be more explicit, but `LVT` / `HVT` for me represent threshold voltages, not the voltage tolerance of the mosfet.

KrzysztofHerman, 2025-03-13:
@smunaut good point. We will discuss internally how to deliver the symbols. Thank you all for this discussion. Definitely we will apply  your suggestions elaborating the symbols for `qucs-s`.

olisnr, 2025-03-20:
@smunaut You are right. the bulk shoudnt be in the middle. but i think it would be better on the side of the S because most times its connected to it.

[image]

smunaut, 2025-03-27:
@KrzysztofHerman Ping. Any idea if/when new xschem symbols will be available ?

sergeiandreyev, 2025-03-28:
Hi @smunaut, the update was part of this PR https://github.com/IHP-GmbH/IHP-Open-PDK/pull/393
It's already merged in `dev` branch, so could you please confirm, and then we can close this issue