IHP-GmbH/IHP-Open-PDK #391: Cant Solve 800x800 µm² coverage ratio DRC error

shafinhossain, 2025-03-10:
My design's layout is of 0.91 mm² and after cleaning the relevant DRCs and density errors, I am still getting these errors which mention the 800x800 µm² coverage ratio.

[image]
[image]
[image]
[image]
[image]
[image]

In a previous issue #337 that I created, one of the contributors mentioned these 800x800 µm² coverage ratio errors could be ignored.

[image]

Is it a problem that can be solved? if not then would it cause any issues in the OpenMPW?.

TIA.

sergeiandreyev, 2025-03-10:
Hi @akrinke, could you please take a look / provide your feedback?

akrinke, 2025-03-10:
Hi @shafinhossain, I think the problem can be explained by the following figure. The DRC calculates the density in 800×800 µm² windows which are shifted by 400 µm to cover the whole chip area. In your case with a (square?) chip with an area of 0.91 mm², there are three windows near the top and right border of the chip, that cover mostly pads. While the density calculation ignores the area outside your chip boundary, the pads often violate the minimum density constraint. I assume that increasing the fill rate in the pad ring solves these issues.

[image]

sergeiandreyev, 2025-03-17:
Hi @shafinhossain, could you please attach the GDS file as a testcase?

shafinhossain, 2025-03-24:
@sergeiandreyev I was able to clean all of the DRCs. Seems like you have to find spaces in per 800x800 um window in order to put all the metal layers and maintain the minimum 25% limit.

Thank you for the cooperation.

sergeiandreyev, 2025-03-24:
good! but please note that having a clean min DRC does not eliminate the possibility of having at the end a non-functional design. In open source DRC deck, we still have a list of rules missing, so please at least go briefly through the DRM, to review chapters that are related to your design, like offGrid (you have circular structures), etc.

sergeiandreyev, 2025-03-25:
Hi @shafinhossain, if there are no more questions on this topic, please let us know if we can close the issue

akrinke, 2025-03-25:
This is a problem in the DRC scripts. The picture I posted above shows the current (incorrect) behavior. The windows in which the density is checked should always be within the chip area. I am working on a fix.