IHP-GmbH/IHP-Open-PDK #392: SRAM: Purpose of BIST pins?

dnltz, 2025-03-10:
https://github.com/IHP-GmbH/IHP-Open-PDK/blob/68eebafcd9b2f5e92c69d37a8d3d90eb266550f5/ihp-sg13g2/libs.ref/sg13g2_sram/verilog/RM_IHPSG13_1P_1024x8_c2_bm_bist.v#L31

What's the purpose of those BIST pins and what's the difference to just ADDR, ...?

smunaut, 2025-03-10:
It's in the doc : https://github.com/IHP-GmbH/IHP-Open-PDK/blob/68eebafcd9b2f5e92c69d37a8d3d90eb266550f5/ihp-sg13g2/libs.ref/sg13g2_sram/doc/RM_IHPSG13_1P_1024x16_c2_bm_bist.txt

Basically there is two ports and which one is used depends on the `BIST_EN` pin state. ( So it's not a dual port sram, more like a mux in front allowing two users ). I think the purpose is to have one used for testing / validation and the other for "runtime".

dnltz, 2025-03-10:
@smunaut thanks :) That makes sense now.

BTW: BIST stands for "Built In Self-Test".

closed 2025-03-10 by dnltz