IHP-GmbH/IHP-Open-PDK #404: Discrepancy between fill area and density computation area causes DRC error.

smunaut, 2025-03-25:
AFAICT:
* The fill algorithm fills everything inside the actual seal ring ( inside the thin metal ring [`39/0`], not the actual `SealRing.boundary` [39/4] ).
* The density computation works over the whole chip area. ( So it's a bit wider, by ~ 36u than where the fill happens )

[image]

The problem we're having is that our chip is 6450 um high. Meaning one of the DRC window starts barely 50u off the top edge. And because of that discrepency, 36u of those 50u are empty (outside the seal area) but they're still taken into account for density calculation meaning reaching the density target is not possible ...).

So we get DRC error for all layers along the top edge.

[image]

( Except for Metal1 because for some reason the seal ring includes a long `Metal1` strip outside the actual seal ring ... not sure why, so for Metal1 we only get the DRC error in the corner, but still same issue ).

sergeiandreyev, 2025-03-25:
Hi @dnltz, @akrinke, could you please take a look at this issue?

dnltz, 2025-03-25:
I'm pretty sure @KrzysztofHerman said there should be not metal fill behinde the sealring. That's why there is not metal fill between 39/4 and 39/0.

akrinke, 2025-03-25:
This is a problem in the DRC scripts. The windows in which the density is checked should always be within the chip area. I am working on a fix.

smunaut, 2025-07-22:
This is fixed by 26a93b74492214cff2ad378f33dba564c8897946

closed 2025-07-22 by smunaut