IHP-GmbH/IHP-Open-PDK #408: Fill pattern generation in Magic has multiple issues.

RTimothyEdwards, 2025-03-25:
I am entering this issue so that it is recorded in the issue tracker and not just a pull request.

The pattern generation parameters in the magic tech file has multiple problems:

1. The density for all metal layers and diffusion is too high, approximately 70% for most layers, while a more reasonable target is 50% or slightly below.
2. The resizing used to eliminate fill shapes that are clipped by keep-out areas is incorrect in a few cases, resulting in some fill shapes either non-rectangular or smaller than allowed.
3. The poly fill is not aligned with the diffusion fill, and in fact is being suppressed by the diffusion fill, resulting in too-low poly density.
4. The density fill script needs rewriting, to handle GDS files as well as .mag files, to automatically find the .magicrc startup script, and to make use of compressed GDS for the pattern fill files, among other things.
5. The density check script has incorrect calculations for the area as counted by whole tiles, having the original 100 count for a 10 x 10 stepped area, which was carried over from the original script used as the source for this one, whereas the steps were changed in this file to step by half the width of a tile, so the total tile area is 4 for a 2 x 2 stepped area.
6. The density check script has incorrect calculations for pro-rating the density of partial tiles on the right and top sides of a layout.

These items are being addressed and will result in a pull request.

mole99, 2025-04-07:
Magic fails fill generation on a design with inductors:

```
Error while reading cell "inductor2$1$1" (byte position 2437336): Unknown layer/datatype in boundary, layer=157 type=0
```

[image]

The layer is named `LBE.drawing`, "For localized back side etch".

RTimothyEdwards, 2025-04-07:
That's not a fatal error;  it's just indicating an error that magic doesn't recognize (and which doesn't affect fill generation).  If magic is "failing fill generation" it's failing for some other reason (although I didn't realize that backside etching was being used for inductors and I should make sure that magic recognizes the layer).