IHP-GmbH/IHP-Open-PDK #410: sram: Missing cells in RM_IHPSG13_1P_1024x8_c2_bm_bist

dnltz, 2025-03-26:
I added RM_IHPSG13_1P_1024x8_c2_bm_bist as SRAM cell. OpenROAD now reports the following error when merging GDSs:

```
[INFO] Checking for missing cell from GDS/OAS...
[ERROR] LEF Cell 'RM_IHPSG13_1P_BITKIT_16x2_EDGE_TB' has no matching GDS/OAS cell. Cell will be empty.
[ERROR] LEF Cell 'RM_IHPSG13_1P_BITKIT_16x2_CORNER' has no matching GDS/OAS cell. Cell will be empty.
[ERROR] LEF Cell 'RM_IHPSG13_1P_BITKIT_16x2_LE_con_corner' has no matching GDS/OAS cell. Cell will be empty.
[ERROR] LEF Cell 'RM_IHPSG13_1P_BITKIT_16x2_LE_con_edge_lr' has no matching GDS/OAS cell. Cell will be empty.
[ERROR] LEF Cell 'RM_IHPSG13_1P_BITKIT_16x2_TAP' has no matching GDS/OAS cell. Cell will be empty.
[ERROR] LEF Cell 'RM_IHPSG13_1P_BITKIT_16x2_LE_con_tap_lr' has no matching GDS/OAS cell. Cell will be empty.
[ERROR] LEF Cell 'RM_IHPSG13_1P_BITKIT_16x2_TAP_LR' has no matching GDS/OAS cell. Cell will be empty.
[ERROR] LEF Cell 'RM_IHPSG13_1P_BITKIT_16x2_POWER_ramtap' has no matching GDS/OAS cell. Cell will be empty.
[INFO] Checking for orphan cell in the final layout...
```

I haven't verified if other sram cells have the same issue.

smunaut, 2025-03-26:
Related to #239 I think.

sergeiandreyev, 2025-03-26:
yes, related to that one, but still, as mentioned in a separate email, these are empty (placeholder) blocks in GDS -> there is nothing expected to be inside, so you have to configure the flow to skip them

dnltz, 2025-03-29:
We can disable that error in openroad with the following export
```
export GDS_ALLOW_EMPTY = RM_IHPSG13_1P_BITKIT_16x2_*
```
That will result in warnings:
```
[INFO] Found GDS_ALLOW_EMPTY variable.
[WARNING] LEF Cell 'RM_IHPSG13_1P_BITKIT_16x2_TAP' ignored. Matches GDS_ALLOW_EMPTY.
[WARNING] LEF Cell 'RM_IHPSG13_1P_BITKIT_16x2_EDGE_TB' ignored. Matches GDS_ALLOW_EMPTY.
[WARNING] LEF Cell 'RM_IHPSG13_1P_BITKIT_16x2_LE_con_edge_lr' ignored. Matches GDS_ALLOW_EMPTY.
[WARNING] LEF Cell 'RM_IHPSG13_1P_BITKIT_16x2_LE_con_tap_lr' ignored. Matches GDS_ALLOW_EMPTY.
[WARNING] LEF Cell 'RM_IHPSG13_1P_BITKIT_16x2_TAP_LR' ignored. Matches GDS_ALLOW_EMPTY.
[WARNING] LEF Cell 'RM_IHPSG13_1P_BITKIT_16x2_POWER_ramtap' ignored. Matches GDS_ALLOW_EMPTY.
[WARNING] LEF Cell 'RM_IHPSG13_1P_BITKIT_16x2_LE_con_corner' ignored. Matches GDS_ALLOW_EMPTY.
[WARNING] LEF Cell 'RM_IHPSG13_1P_BITKIT_16x2_CORNER' ignored. Matches GDS_ALLOW_EMPTY.
[INFO] Checking for orphan cell in the final layout...
```

@sergeiandreyev @KrzysztofHerman I would like to add that to the ihp-sg13g2 platform config in ORFS. Do you see any problems with that?

sergeiandreyev, 2025-03-29:
Hi @dnltz , great, thank you! From my side no objections, the only thing - we need to verify the empty GDS cells in all the SRAM macros provided so far, if the set pattern will cover all of them

sergeiandreyev, 2025-03-29:
and on the other side, that it will not capture the cells which are not supposed to be empty

dnltz, 2025-04-14:
@sergeiandreyev I would propose to add following to ORFS:

```
export GDS_ALLOW_EMPTY = RM_IHPSG13_1P_BITKIT_16x2_(TAP|EDGE_TB|LE_con_edge_lr|LE_con_tap_lr|TAP_LR|POWER_ramtap|LE_con_corner|CORNER)
```
That should only disable thos empty cells in SRAM macros.

closed 2025-04-17 by dnltz