IHP-GmbH/IHP-Open-PDK #412: Angle checks

mole99, 2025-03-27:
I'd like to make a doodle for some chip art, and I'm wondering if I can use non-manhatten geometry on TopMetal2.

In the layout rules it is stated under 3.1 Grid Rules:

> Following layers are only allowed on 90, 135, 180, 225, and 270 degree angles: GatPoly, Activ, Metal1, Metal2, Metal3, Metal4, Metal5, TopMetal1, TopMetal2

Well, if I interpret that correctly that means in addition to 90° angles I can also do 45° angles on TopMetal2 (or 135°, depends how you look at it). (Rule `Gat.f` also uses 45° to describe the angle, so this seems to be inconsistent with the grid rules.)

But if possible, I would like to do smooth curves.

I've seen traces with round edges in one of the user submitted designs for the April tapeout for multi-GHz stuff. It's this design here: https://github.com/IHP-GmbH/TO_Apr2025/pull/4

[image]

The traces with the round edges are on TopMetal2. As it is, this would violate the angle rules, but both the KLayout minimum and maximum rule set don't catch it.

So is this an error in the documentation, the DRC set, or the design?

If technically feasible, I would love to make smooth curves on TopMetal2 ;)

mole99, 2025-03-27:
Hi @sergeiandreyev, thanks for your reply!

I don't think the layout violates the offgrid rules. The grid is 5nm and the coordinates for all points of the curve are on the 5nm grid.

So the only issue would be the `non45R_TopMetal2` rule.
Is there a reason why TopMetal2 is restricted to these angles? I think for sub-GHz it's important to have round traces.

If that's not possible, you should probably let the person behind this PR know that their design can't be manufactured: https://github.com/IHP-GmbH/TO_Apr2025/pull/4

akrinke, 2025-03-28:
@sergeiandreyev `OffGrid` rules are part of the Maximal Rule Set and also listed in `README_maximal.md`. This includes a check for `TopMetal2` (`OffGrid.TopMetal2`). The DRC checks the position of individual vertices. Since GDS doesn't support round features, curves have to be approximated. Interestingly, and as @mole99 also wrote, the layout from the first comment does not contain OffGrid errors!

Currently, we do not check for 45 degree angles, so rules like `non45R_TopMetal2` are missing. We will add these rules.

mole99, 2025-03-28:
Thanks @akrinke for confirming that there are no offgrid errors in the design.

Still, I don't really see the reason for the `non45R_TopMetal2` rule.

I found this paper of a transmitter/receiver using the SG13G2 process: https://www.ihp-microelectronics.com/php_scripts/publications/full_text_final_files/schmalz-schmalz-ieee_j.-microwaves-2-582-2022-2022.pdf

One can clearly see that the antenna uses round geometry, why shouldn't this be possible with the open source PDK?

sergeiandreyev, 2025-03-28:
> [@sergeiandreyev](https://github.com/sergeiandreyev) `OffGrid` rules are part of the Maximal Rule Set and also listed in `README_maximal.md`. This includes a check for `TopMetal2` [...]

Hi @akrinke, yes, sorry, somehow I didn't catch them.. I had a discussion internally on this topic, there should be no `offGrid` LVS violations on circular / round Metal shapes (it means we have to do some fixes internally), even if they occur they are of very low priority, so their impact on manufacturability is minimal
copying @stefansimon42 :)
@mole99, could you please attach a testcase, that we can cross check internally

sergeiandreyev, 2025-03-28:
> Still, I don't really see the reason for the non45R_TopMetal2 rule.
on this we have also a disconnect which we should resolve internally

sergeiandreyev, 2025-03-28:
Hi @Atif07Ali, Angle rules are not part of short (min) DRC, so will pass rejection tests, no worries here

mole99, 2025-03-28:
> Hi [@Atif07Ali](https://github.com/Atif07Ali), Angle rules are not part of short (min) DRC, so will pass rejection tests, no worries here

Thanks @sergeiandreyev, that's exactly what I wanted to know.

Just to confirm, this means I can use non-45° angles on TopMetal2 (for some chip art) for the April tapeout and the design won't be rejected (as long as everything is on-grid of course).

Perhaps it would be good to update the documentation and clarify that the angle rules are not a manufacturing requirement, but rather a recommendation :)

sergeiandreyev, 2025-03-28:
> Perhaps it would be good to update the documentation and clarify that the angle rules are not a manufacturing requirement, but rather a recommendation :)

I'll let @stefansimon42 to comment on this