IHP-GmbH/IHP-Open-PDK #416: DRC shows spurious MnFil.a1 and TMnFil.a violations

smunaut, 2025-03-29:
So I was trying Magic fill algorithm and then running the result through DRC and there was a lot of violations reported.

```
Rule M1Fil.a1: 55502 error(s)
Rule M2Fil.a1: 81040 error(s)
Rule M3Fil.a1: 179170 error(s)
Rule M4Fil.a1: 197511 error(s)
Rule M5Fil.a1: 182144 error(s)
Rule TM1Fil.a: 1735 error(s)
Rule TM2Fil.a: 66799 error(s)
```

However when loading the resulting DRC and inspecting it, the violations appear to be spurious. Here's an example for `M1Fil.a1` :

[image]

The edge pair don't belong to the same fill polygon at all ... and those polygons are far enough appart and are wider than 1u.

Attached a reproducer. It's only the fill pattern itself without the design, but it's enough to see the issue.
There is also a minimized lydrc file with just the `M1Fil.a1` rule to run faster but you can also use the maximal DRC to reproduce.
( Sorry for the .tar.xz inside a .zip but I needed xz to be under file limit and github only accepts zip ... )

[drc_bug.zip](https://github.com/user-attachments/files/19516735/drc_bug.zip)

(cc @RTimothyEdwards )

sergeiandreyev, 2025-03-30:
Hi @akrinke, could you please take a look?

akrinke, 2025-03-31:
Hi @smunaut, thank you for the report! This is a definitely bug in our DRC scripts. I am working on a fix right now.

akrinke, 2025-04-04:
I've created PR https://github.com/IHP-GmbH/IHP-Open-PDK/pull/432 that fixes this issue.

KrzysztofHerman, 2025-04-15:
@smunaut @akrinke solved in #432

PR #432 (2025-04-15): Update KLayout DRC scripts: Fix bug in width checks
- width check produced false positives for edges of different polygons
- fixes #416
commit 0b39dba9 (2025-04-15): Fix bug in width checks

- width check produced false positives for edges of different polygons
- fixes #416
closed 2025-04-15 by KrzysztofHerman