IHP-GmbH/IHP-Open-PDK #418: Device Present in Netlist but not showing during LVS

rupokdas147, 2025-03-30:
Capacitor **CDC1** is present in netlist but not showing in the reference during LVS check.

[image]

[image]

d-m-bailey, 2025-03-31:
@rupokdas147 In the source side, `CDC1` and `CDC3` have been combined to `DC3` (check the sizes).

On the layout side, the devices `$6` and `$2` have not been combined. Can you verify that `$6` and `$2` are parallel (connected the same) in the layout?

There may be a problem with the reduction rules. It appears that `w`, `l`, and `A` are all summed, while `P` remains the same.

rupokdas147, 2025-03-31:
@d-m-bailey yes they are parallel in the layout side. This is the netlist provided by Klayout.

[image]

sergeiandreyev, 2025-04-02:
Hi @atorkmabrains, @FaragElsayed2, could you please take a look?

closed 2026-03-11 by rupokdas147