IHP-GmbH/IHP-Open-PDK #420: LVS extracted netlist for npn13G2 with NE=2 also writes m=2

olisnr, 2025-03-31:
if i LVS a circtuit with `npn13G2 NE=2` the extracted netlist will write the line:
```
Q$26 \$65 in_p \$117 \$1 npn13G2 AE=0.063p PE=1.94u AB=12.988p PB=15.62u
+ AC=12.976684p PC=15.61u NE=2 m=2
```
is this correct? or a problem?

sergeiandreyev, 2025-04-03:
Hi @olisnr, for now I can say that the above is inconsistent with the implementation in commercial PDK, while if it is a bug or not - this has to be analyzed..
Is the LVS clean in KLayout when you check it?
@atorkmabrains, @FaragElsayed2, could you please take a look?

sergeiandreyev, 2025-04-08:
well, it seems is a bug

KrzysztofHerman, 2026-03-11:
Hi @olisnr, PRs #800, #828, and #862 have been merged. Do these changes resolve the issue? If yes, please consider closing this issue. Thanks!