IHP-GmbH/IHP-Open-PDK #421: sub.sym: more description please

EngGhaith, 2025-04-01:
Hi everybody,
about the PDK symbol sub.sym in Xschem:
Should we tie the sub.sym to GND, or should we use this symbol instead of the Xschem-original gnd.sym? or both are equivalent?

Tank you very much.
Ghaith.

KrzysztofHerman, 2025-04-01:
@EngGhaith 
The purpose of the symbol is to annotate in the netlist a global net named `sub!`. The net (and the associated node) represents substrate so you can use it whenever you want to keep devices connected to the substrate. It is useful if you would like to distinguish between VSS potential rail and bulk potential and make use of `ptap1` device. Some other devices like 'isolbox', `dantenna`, `ptap1`   also connect directly to the substrate.   

[image]

This is the netlist corresponding  to the schematic show above.
The `.GLOBAL` qualifier will make the `sub!` net available at each level of hierarchy 
```
XD1 sub! net1 dantenna l=0.78u w=0.78u
XR1 sub! net2 ptap1 R=262.8 w=0.78e-6 l=0.78e-6
XD2 net3 net4 sub! isolbox l=3.0u w=3.0u
.GLOBAL sub!
```

EngGhaith, 2025-04-01:
Is there any specian devices, which must been manually tied to sub?
e.g. substrate of NPN HBT? Or they are already tied to substrate?

KrzysztofHerman, 2025-04-01:
@EngGhaith 
In general the fourth terminal is a bulk (substrate terminal) and you can connect it to the `sub!` net or `VSS` net. If you connect to VSS you neglect the finite resistance between the substrate and `VSS` which you can model using `ptap1` device. 
The exception are our `rfmos` devices which already have a tap ring  (which act as `ptap1` device)  and its resistance is a part of the model so it is justified to connect the bulk directly to the `VSS` on the schematic.