IHP-GmbH/IHP-Open-PDK #422: Capacitors for RF: more documentation needed

EngGhaith, 2025-04-01:
Hi everybody,

could you please provide informations about useful RF-Caps?
In Xschem there are 3 Symbols: cap_cmim.sym, cap_rfcmim.sym, cap_cpara.sym.
I wanna know which is more accurate and has less Process Variation (PV), because I am using them in Matching Circuits.
CMIM is documented in the process spec file, but what about RFCMIM and CPARA?
What is the difference between CMIM and RFCMIM?

Typically CMOM is used for RF, due to its good modelling and low PV. Do you have such cells?

Thank you in advance.
Ghaith.

VolkerMuehlhaus, 2025-04-02:
cap_rfcmim is an extended version of cap_cmim, with more layout restrictions (feedlines from opposite sides, additional ground frame on lower metals) and with a more accurate RF model. This model includes series inductance due to physical size (-> SRF) and losses, whereas the standard cap_rfcmim is modelled more or less as an ideal capacitor. Regarding the C value calculated at low frequency, both models are identical.

EngGhaith, 2025-04-03:
@VolkerMuehlhaus Thank you very much. It seems to be a good modeled Cap for RF now.

I have the following questions about RFCMIM in Xschem:

[image]

Where should we tie the pin bn?
This symbol is typically used for polarized caps, ist this cap polarized?
What is the impact of the wfeed? how to choose it well? depends that on operating frequency or current density?

And maybe also in KLayout:

[image]

What should we manipulate?
I mean ofcourse w, l and wfeed. THen, C should be automatically calculated, or? because I have to enter it manually to the field (according to that calculated one in the schematics).
What about other parameters (Cspec, Wmin, Lmin, Cmax)?

VolkerMuehlhaus, 2025-04-03:
@EngGhaith 
The bn is the bulk node, it must be connected to substrate potential. This path in the underlying circuit model is used for parasitics to the substrate.
rfcmim is not polarized, it's essentially a plate capacitor.
wfeed is the feed line width, is has some effect on parasitic series inductance and shunt capacitance to the substrate.

I can't comment on calculation within the Klayout form, if that is perhaps missing. I only know this model from the Cadence flow because I created the underlying circuit model and equations.

VolkerMuehlhaus, 2025-04-03:
Yes, connecting the bn terminal to sub! is correct.

sergeiandreyev, 2025-10-07:
Hi @EngGhaith,
> Then, C should be automatically calculated, or? because I have to enter it manually to the field (according to that calculated one in the schematics).
What about other parameters (Cspec, Wmin, Lmin, Cmax)?

All these parameters should be automatically recalculated in KLayout Pcells, the thing is the callbacks are still missing - we're working on this..