IHP-GmbH/IHP-Open-PDK #424: Metal1.filler in sealring

mole99, 2025-04-02:
This commit added `Metal1.filler` under the `Passiv.drawing` of the sealring: https://github.com/IHP-GmbH/IHP-Open-PDK/commit/516b01c3a0f7a7a1d6e0dfe2875fceddd97c84e7

This is a violation of rule `M1Fil.a2`, as its side is much larger than 5um.

@sergeiandreyev Do you remember why this change was made? Perhaps to fix the density violations of a specific design?

sergeiandreyev, 2025-04-02:
Hi @mole99, this update was a result of internal discussion in PDK group (based on feedback from Mask team?) during the last year tapeout in Autumn. I checked my message history, and this violation I've already noted and forwarded at the time of the update (it is also caught by commercial tools). Unfortunately, it seems it didn't move from that point. I will restart the discussion. Thank you for reminder!

mole99, 2025-04-03:
Thanks @sergeiandreyev, does this mean I should keep or remove the `Metal1.filler` from the sealring for the April tapeout, or is there not much difference?

sergeiandreyev, 2025-04-03:
Hi @mole99, I would say you should waive this violation, maybe @KrzysztofHerman has some updates on the topic..
btw, how did you catch it?

mole99, 2025-04-03:
Well, I've already removed it, so if it's not too much of a problem, I'll leave it like that :)
I saw that Sylvain wrote about it in FOSSi chat, he also removed it from the Tiny Tapeout chips.

sergeiandreyev, 2025-04-03:
ok.. this rule is not part of the rejection test, so the submission check will not fail
the question is how you found this violation, this rule is not yet implemented in KLayout DRC, am I right @akrinke?

smunaut, 2025-04-03:
It's implemented in Magic yes.
I've also got it in Calibre.

akrinke, 2025-04-07:
@sergeiandreyev Rule `M1Fil.a2` is not implemented in the KLayout DRC script, yet. I am working on an update.

sergeiandreyev, 2025-04-08:
Hi @mole99, it was indeed introduced in trying to optimize the filling in KLayout, but it was incorrect - so I've decided to roll back / remove this update (https://github.com/IHP-GmbH/IHP-Open-PDK/commit/845ecd1dc4ed31ba269bc329e761b77070a42289), we should focus on enhancing the filler scripts.

closed 2025-08-15 by sergeiandreyev