IHP-GmbH/IHP-Open-PDK #425: resistor warning from commercial PDK

olisnr, 2025-04-02:
if somebody with the commercial PDK looks at my .GDS there are warnings in the netlist:
```
***************************************
.SUBCKT rppd$1
** N=4 EP=0 IP=0 FDC=0
** WARNING: BAD DEVICE on layer Rppd_nwell at location (0,0) in cell rppd$1 (see extraction report).
*.CALIBRE WARNING BADDEV BAD DEVICE(s) detected by extraction in this cell. See extraction report for details.
.ENDS
***************************************
```
is this a problem or normal?

olisnr, 2025-04-03:
[VCO_HBT_inv_13.gds.zip](https://github.com/user-attachments/files/19592293/VCO_HBT_inv_13.gds.zip)

its from this inverter

olisnr, 2025-05-20:
@sergeiandreyev do You know something new about the RPPD?

d-m-bailey, 2026-03-04:
@FaragElsayed2 Quick question.

Is it necessary to flatten the gds? 

What about just unchecking the `Cell Content 🔲 Save Pcell and library context information (format specific)` in the klayout save dialog?