IHP-GmbH/IHP-Open-PDK #431: OpenLane support

jadoro, 2025-04-04:
Hello,

I was trying to synthesisze a small test deisgn for `ihp-sg13g2` using OpenLane and found that currently (68eebaf), no `openlane/config.tcl` is provided in `libs.tech`.
Digging a bit, I found [this IHP press release](https://www.ihp-microelectronics.com/services/research-and-prototyping-service/fast-design-enablement/open-source-pdk), stating
> Within the project there will be two established open source digital design flows that can be used with the IHP Open Source PDK. While the OpenROAD flow is already supported, the setup for the OpenLane flow will be provided within 2024.

I just wanted to kindly ask whether there is any update on this statement, especially given the [unclear future of efabless](https://web.archive.org/web/20250324223745/https://efabless.com/notice), who seemed to be a major driving force behind OpenLane.

Thanks a lot for your efforts!

KrzysztofHerman, 2025-04-04:
@jadoro there is ongoing WiP to have support for openlane you can try it out using [this](https://github.com/mole99/IHP-Open-PDK/tree/openlane) fork & branch. Maybe @mole99 could give you some more details.

jadoro, 2025-04-04:
@KrzysztofHerman That looks great, thanks a lot for the hint! I'll give it a try.

KrzysztofHerman, 2025-04-04:
@jadoro give us some feedback about the performance and issues if you find any.

mole99, 2025-04-04:
I have a few examples ready to run: https://github.com/mole99/ihp-openlane-examples
(Note, the configs may need to be adjusted, i.e. the die area for ttihp0p2-one-sprite-pony)

For simple macros it already works fine, but there are still a few issues that need to be sorted out.
I've been designing a chip for the April tapeout using OpenLane, and there were a few issues for which fixes still need to be properly integrated. This is going to happen during the next few months.

jadoro, 2025-04-04:
Thanks a lot, this looks like a really nice example to get started with!

We're currently using OpenLane 1, which seems to expect slightly different variables in `config.tcl` (e.g. `CELLS_LEF` instead of `CELL_LEFS`, `LIB_SYNTH` being defined, …). I will probably try to see if I can achieve OL1 compatibility in reasonable time, otherwise me might migrate our flow.

mole99, 2025-04-04:
You're welcome! Yes, many of the variables have been renamed or deprecated since OpenLane 2.

I would recommend you to upgrade to OpenLane 2 as it has a lot of improvements compared to OL1 and continues to be developed. I also enjoy using Nix instead of Docker to manage the tools and their environment.

sergeiandreyev, 2025-11-03:
Hi @jadoro, could you please try latest `dev` branch of the PDK with added LibreLane support and let us know how it goes?