IHP-GmbH/IHP-Open-PDK #435: `tt_stat` models are picking new random parameters per device instance

smunaut, 2025-04-07:
So what is expected when running a `tt_stat` sims, is that random device parameters are picked once (per device type) and then applied to all instances of that devices in the design.

This is not what happening. Instead each instance gets different random parameters which as you can imagine is a bit unrealistic because the spread from `tt_stat` is quite wide and using random parameters in current mirrors or differential pairs leads to "interesting" results.

The way the models pick the parameters is using `.param xxxx = gauss(....)` and if you believe the `ngspice` manual, theses should be evaluated once at startup and the same value used everywhere, which is exactly what you'd want here.

Unfortunately, the manual is wrong ... I've opened a thread on the `ngspice` mailing list ( https://sourceforge.net/p/ngspice/discussion/133842/thread/c13bbd10c6/ ) but the gist of it is that there seem to be a deliberate, undocumented behavior that any `param` that has a statistical distribution function in it is re-evaluated at each use.

That obviously wreaks havoc here the way the models are written ... it basically creates a mismatch simulation but with a much greater spread of parameters than you'd want for a mismatch sim :/ 

One work around I've tried is instead of : 
```
.param xxx = gauss(...)
```

is to use :

```
.param rnd_xxx = gauss(...)
.param xxx = rnd_xxx
```

Then `xxx` will actually have a stable value.

This is an example patch that applies that work around to the models here.
[ihp_stat_workaround.diff.gz](https://github.com/user-attachments/files/19628464/ihp_stat_workaround.diff.gz)

Note that this is for testing/evaluation only. I only checked the `lv nmos` with it applied and I'd also like to see how the `ngspice` thread before applying anything, but I'm opening the issue here to keep track / not forget about it.

dwarning, 2025-04-07:
@smunaut 
I think something was confused. If we speak about tt_stat the model parameters are varied - not instance parameters. The variation of all devices with same model should give similar range as we have for corner simulation, but randomly. This approach should not mixed with instance variations which give us the mismatch result between two (or more) devices of same model type.

Can you prepare a simple test case where you show the failed behavior.

I attached my test case based on the actual state of dev branch.

[lv_nmos_id_vd_vg_stat.txt](https://github.com/user-attachments/files/19632516/lv_nmos_id_vd_vg_stat.txt)

Using that I get this:

[image]

what looks plausible for me.

smunaut, 2025-04-07:
I understand that `tt_stat` is meant to vary model parameters and not instance parameters. But that's not what it _actually_ does because `ngspice` doesn't behave like the manual says it does. You could very well call this an ngspice bug but the result is the same, ATM `tt_stat` doesn't do what it should.

Here's a minimal test case.
It's two identical transistors, fixed Vgs and Vds, measuring Id of each. From run to run the Id value should vary but be the same for the two transistors (they are _identical_) .... but that's not what happens. Each transistor gets a different Id.

[tt_stat_test.txt](https://github.com/user-attachments/files/19632975/tt_stat_test.txt)

dwarning, 2025-04-07:
OK, now I understand and can confirm. There goes something wrong with ngspice. Must see what the outcome is from Holger's investigation.
An alternative way to your recommendation with "re-parametrization" could be to shift the gauss() functions direct to the model parameter set instead. Will try it at the next opportunity.

dwarning, 2026-11-03:
Solved by:
#435 workaround for statistical simulations - provided by user smunaut

PR #441 (2025-05-07): #435 workaround for statistical simulations - provided by user smunaut
Fixes #435

This patch fixes the weakness of ngspice to handle statistical variations in model parameters which are include inside subckts.
Better solution would be to prevent usage of instance parameters in model parameter calculation and place the model definition outside the subckt.
closed 2026-11-03 by dwarning