IHP-GmbH/IHP-Open-PDK #436: DRC density error M1.j/k-M5.j/k

joerdsonsilva, 2025-04-07:
When running the minimum DRC in Klayout, these density errors appear, even with fillers. 

[image]

olisnr, 2025-04-07:
i dont see the image, but i had the MFil.h/k errors. i think it was from the PADs and the sealring. i did make a 222x555 vias PyCELL over all layers in the region whit the problem, and it was gone.

joerdsonsilva, 2025-04-07:
Thanks for the feedback. I'll test something like this. I saw someone reporting something similar to this density error in the slack group, but I don't know if it's something that can be solved with script or layout.

(https://open-source-silicon.slack.com/archives/C0549N94FD3/p1736933192511219)

olisnr, 2025-04-08:
the density error message should include the measured density, because now You arent sure if You need more or less metal.

akrinke, 2025-04-08:
@olisnr I agree, but I don't know how to annotate DRC errors in KLayout. I think there is a separate script that calculates the density.

olisnr, 2025-04-08:
@akrinke the script only calculates the global density, not the local one. You wont see this error

joerdsonsilva, 2025-04-09:
The issue was fixed by editing the sg13g2_filler_Metal.lym script. By editing the "distance_m1/5" values ​​to 1.0, DRC did not report the density error.

closed 2025-04-10 by joerdsonsilva