IHP-GmbH/IHP-Open-PDK #437: sealring PyCELL dont work on KLayout 3.0

olisnr, 2025-04-07:
only a message for the future...

danielh186, 2025-04-10:
I also just ran into this issue. Sealring generation will not work with klayout version 0.30.0 and also opening and resaving a perfectly fine gds will totally destroy the sealring.

Klayout throws the following error message:
```
 ERROR: Traceback (most recent call last):
   File "/home/daniel/IHP-Open-PDK/ihp-sg13g2/libs.tech/klayout/tech/pymacros/../../python/pycell4klayout-api/source/python/cni/dlo.py", line 225, in produce
     self._impl.genLayout()
   File "/home/daniel/IHP-Open-PDK/ihp-sg13g2/libs.tech/klayout/python/sg13g2_pycell_lib/ihp/sealring_code.py", line 152, in genLayout
     ihpCopyFig(groupId, Point(l, w), 'R180')
   File "/home/daniel/IHP-Open-PDK/ihp-sg13g2/libs.tech/klayout/python/sg13g2_pycell_lib/ihp/geometry.py", line 1186, in ihpCopyFig
     cons(newList, dbCopyShape(item, pnt, rot))
   File "/home/daniel/IHP-Open-PDK/ihp-sg13g2/libs.tech/klayout/python/sg13g2_pycell_lib/ihp/geometry.py", line 187, in dbCopyShape
     dbMoveFig(copyId, pnt, rot)
   File "/home/daniel/IHP-Open-PDK/ihp-sg13g2/libs.tech/klayout/python/sg13g2_pycell_lib/ihp/geometry.py", line 199, in dbMoveFig
     fig.moveBy(pnt.x, pnt.y)
   File "/home/daniel/IHP-Open-PDK/ihp-sg13g2/libs.tech/klayout/tech/pymacros/../../python/pycell4klayout-api/source/python/cni/polygon.py", line 75, in moveBy
     dataBaseUnits).to_simple_polygon().to_dtype(Tech.get(Tech.techInUse).dataBaseUnits)
 AttributeError: 'SimplePolygon' object has no attribute 'to_simple_polygon'
```

ThomasZecha, 2025-04-11:
Fixed by https://github.com/IHP-GmbH/pycell4klayout-api/commit/4c463c43991fb0967a824906518b66047bfb33c4

PR #446 (2025-05-06): PyCell for Klayout
-updated submodule pycell4klayout-api

Fixes #437