IHP-GmbH/IHP-Open-PDK #438: Relax metal filler rules

mole99, 2025-04-08:
We have seen in the previous tapeouts that fill generation still proves to be a challenge. This may be partly due to the current fill generation scripts, which do not work equally well for every layout. In part, it may also be due to the metal filler rules, which are unnecessarily restrictive.

Especially with digital designs, fill generation for the metal layers can be very painful. Each routing layer has a preferred direction, which makes it difficult to insert square filler shapes.

Take this Metal2 section of my latest design for example:

[image]

It mostly consists of horizontal routing which makes it hard to insert square filler, but it's not dense enough to meet the local density limit by itself.

I had to make five passes of the smallest fill shape with different offsets to get it to meet the density limit. This costs a lot of runtime, but is probably not necessary for most designs.

The smallest fill for Metal1-5 of 1x1um with 0.42um spacing seems overly conservative. In comparison, the smallest metal fill on sky130 is 0.3x0.3um with 0.2um spacing. I'm aware that the metal widths and spacings for the two processes are different, still it seems too restrictive.

Perhaps the following change could be considered: 

1. Lower the minimum Metal1-5 filler width to 0.5 (`MFil.a1`)
2. Create a new rule to set the minimum filler area to 1um²

This should keep the risk of delamination about the same (?), but would allow fill shapes such as 0.5x2um or 2x0.5um that can be inserted into digital designs much more easily.

dnltz, 2025-04-10:
If we have smaller Mx filler rules, we could define three cell sizes:

- S: 0.5
- M: 1.42
- L: 3.26

each is times 2 bigger with a distance of `0.42` included. That means we could start filling L. Afterwards, add M where L doesn't fit and the same with S.

mole99, 2025-04-10:
Yes, that's generally a good idea. In fact, magic already does three passes for each metal layer with different sizes. With the proposed minimum metal width, this would probably work for most, if not all, layouts.