IHP-GmbH/IHP-Open-PDK #439: OpenEMS TLine Example and TLines in SG13G2

EngGhaith, 2025-04-08:
Hi Everybody!
In the example you have provided about simulating TLines with openEMS I have observed, that there is a SiO2 above the top line Metal:

[image]

Is that right modeled?
This SiO2 would change the behaviour of simple microstrip to embedded microstrip.

Can we get an etched SiO2 (Passivation/Air above TM2) above th Line in your opensource PDK and how to do it in the GDS?

Thank you in Advance.
Ghaith Al Sabagh.

VolkerMuehlhaus, 2025-04-08:
Let me add one detail: In the process specification, you can find a drawing of the actual chip cross section (as manufactured) which has a **conformal** layer of 1.5µm oxide and 0.4 SiN on top. The planarized model is only a simplification for EM simulation, and will indeed over-estimate the capacitance a bit.

The 0.4µm SiN layer can be removed/dismissed to speed up openEMS simulation, if all the other mesh dimensions in your model are much larger. As you might know, the timestep in FDTD is determined from the smallest mesh cell dimension, so we don't want to have a small mesh dimension if it's not relevant anyway. That's why SiN is missing in the stackup/example that you used.

That said, the SiO2 thickness in your screenshot looks a bit large, compared to the 3µm TM2 layer. The oxide should be 1.5µm above TM2, but it's actually 3.5µm in the screenshot that you show. There is something wrong with the stackup in this example - shame on me, I had created the code and need to check where this extra 2µm results from!

VolkerMuehlhaus, 2025-04-08:
@KrzysztofHerman 

Hi Krzysztof, can you fix this mistake for me? The code says 

`   SiO2_thick = 17.73`

but I just double checked that the correct value would be 

`   SiO2_thick = 15.73`

which is 2µm less indeed. That's why the oxide above TM2 looks so thick in the screenshow shown by @EngGhaith.

The wrong oxide thickness value is in all 3 *.py models here:
https://github.com/IHP-GmbH/IHP-Open-PDK/tree/main/ihp-sg13g2/libs.tech/openems/testcase/SG13_Octagon_L2n0/OpenEMS_Python
and in all *.m models here:
https://github.com/IHP-GmbH/IHP-Open-PDK/tree/main/ihp-sg13g2/libs.tech/openems/testcase/SG13_Octagon_L2n0/OpenEMS_Matlab

KrzysztofHerman, 2025-04-14:
> Hi Krzysztof, can you fix this mistake for me? The code says

Hi @VolkerMuehlhaus I  have just created a PR #450.  Could you please confirm that it fixes the issue ?

VolkerMuehlhaus, 2025-04-22:
> and calculated the impedance of the line:

@EngGhaith Sorry for my late reply! Something is wrong with your line impedance calculation, from S11 < -25dB we can see that line impedance is really close to 50 Ohm. You should check your math with some known data.

If you have calculated the 2-port data for your line, you might also use my RLGC line extraction code here:
https://github.com/VolkerMuehlhaus/lumpedmodel/tree/main/rlgc_from_s2p
which calculates transmission line model parameters at one single frequency, and also shows the line impedance.

If you look at the code there, you will notice this simple method of calculating Zline:

`Zline = np.sqrt(z11/y11)`

where np.sqrt is the sqrt() function from numpy library.

KrzysztofHerman, 2025-08-15:
@VolkerMuehlhaus @EngGhaith 
I have reviewed this issue once again. From the PDK point of we have merged the #450 hot-fix. Is there anything else to fix  or can we close.

EngGhaith, 2025-08-15:
From my side nothing more to do.

closed 2025-08-15 by EngGhaith