IHP-GmbH/IHP-Open-PDK #443: Klayout LVS of GDS vs two different CDL netlists is clean

b10346, 2025-04-10:
Hello,

It is found strange behavior of LVS deck in Klayout version 29.12.
Two CDL netlists (cdl_v1 and cdl_v2) have a difference in a pins (B1 and B2) to gates connection (see attached files).
Klayout LVS shows "Clean" results with the same GDS layout.

Merging of stacked MOS structure can cause potential errors in circuits at clean LVS results.

[image]
[image]
[image]

smunaut, 2025-04-11:
But what does the LVS report say ?
AFAIR KLayout won't necessarily match pin names, it'd just say that B1 of netlist = B2 of layout and vice-versa in the report ?

sergeiandreyev, 2025-04-11:
[image]

d-m-bailey, 2025-05-19:
@b10346 @sergeiandreyev This issue is currently being [discussed here](https://github.com/IHP-GmbH/IHP-Open-PDK/issues/484).

Most cases should be detectable with a rule modification. I think missing pins is currently undetectable, though.

FaragElsayed2, 2025-08-28:
To be covered in #644