IHP-GmbH/IHP-Open-PDK #452: How to connect the substrate contact of HBT NPN to substrate?

EngGhaith, 2025-04-15:
I have this very simple NPN:

[image]

And I have this in the LVS:

[image]

In Xschem it is connected like that:

[image]

Thank you very much.
Ghaith Al Sabagh.

KrzysztofHerman, 2025-04-15:
@EngGhaith 

please review the examples here
`$PDK_ROOT/ihp-sg13g2/libs.tech/klayout/tech/lvs/testing/testcases/unit/bjt_devices`

sergeiandreyev, 2025-05-09:
Hi @EngGhaith, did you find needed information in the examples?

EngGhaith, 2025-05-09:
Actually I have found, that the device is already connected to substrate via the psD and active ringe around it.
I have tried to do it manually by drawing in the invisible substrate layer.
I think it would be better, to remove the terminal from Xschem-Symbol and replace it with Sub-drawing (Also only triangle with label sub!, withoit terminal)
You would get a 3 terminal device. 
Then you could use Design rules to ensure that the device is connected to substrate.

From my point of view: it makes non sense to make a 4 terminal device, if you will force the designer to connect the S-terminal to sub! to have the guarsnteed performance of the device.

Thank you very much.
Ghaith.