IHP-GmbH/IHP-Open-PDK #456: Inductor PyCells not recognized by LVS

akrinke, 2025-04-16:
Inductor2/3 PyCells are currently not recognized by LVS. The PyCells create pins `LA`, `LB`, and `LC` on `TopMetal1.pin` and `TopMetal2.pin`. The LVS looks for pins on `IND.pin`. Creating copies of the pin rectangles on this layer solves the problem.

Maybe a regression test that performs LVS on a layout that contains PyCell instances can be added to the PDK.

sergeiandreyev, 2025-04-16:
Hi @akrinke, I confirm - we have a disconnect in the `inductor` device layout, this should be fixed on Pycell side. Thanks for finding!

sergeiandreyev, 2025-04-16:
we have LVS regression tests, but these were developed when we had only about half of the device Pycells available, so these tests are checking the device layouts from `sg13g2_pr`..

sergeiandreyev, 2025-05-17:
Hi @akrinke, I've fixed the `inductors` Pycells, could you please check if now they work on your side?

akrinke, 2025-08-06:
Yes, the `inductor2` and `inductor3` PyCells are now extracted correctly.

closed 2025-08-06 by akrinke