IHP-GmbH/IHP-Open-PDK #457: Question about layouting rfcmim

EngGhaith, 2025-04-16:
Hi!
I have this cap:

[image]

First: I am allowed to flip the cap horizontally like that, or?
Second: Should I tie the contact TIE to substrate or it is already done automatically by the PCell?

Thank you very much.
Ghaith.

EngGhaith, 2025-04-17:
Another question:

[image]

Are we allowed to route something under the cap? (between Metal5 and the guard ring on metal 1)?
What are these layers Metalx.noqrc?

VolkerMuehlhaus, 2025-04-22:
I recommend to **rotate** 180° instead of flipping.
The pcell physically connects the ground frame to the substrate. In schematic, you need to connect that node to substrate.

Metalx.noqrc excludes the device area from parasitic extraction, because the circuit model already includes all parasitics.
To have a valid electrical model, you must **not** route other signals under the rfcmim.

KrzysztofHerman, 2025-04-25:
@EngGhaith 
I confirm the recommendation of @VolkerMuehlhaus to rotate 180 deg. 
I also confirm the second suggestion about the bulk node. You can use the `sub` symbol which, will annotate a global net sub! in the netlist. If you do not have further questions please close this issue.

closed 2025-04-25 by EngGhaith