IHP-GmbH/IHP-Open-PDK #459: LVS Problem of two in series connected capacitor

EngGhaith, 2025-04-22:
Hi
I have two in series connected RFCMIMs and KLayout throws me this error

[image]

Why?
Layout:

[image]

Schematic:

[image]

BW, Ghaith.

olisnr, 2025-04-23:
may be there is a problem with an antenna-effect?

EngGhaith, 2025-04-23:
> may be there is a problem with an antenna-effect?

IDK ...
DRCs are clean...

akrinke, 2025-04-23:
The DRC does not currently check for antenna effects. I have no idea how an antenna effect could affect LVS since it shouldn't create an electrical connection.

olisnr, 2025-04-23:
@akrinke it could be, that big floating metal area get to big voltages in production for the MiM isolation. but i dont find any specification in the layout rules, therefore it was just an unfounded assumption...

akrinke, 2025-04-23:
@olisnr Yes, I agree – I just wouldn't expect it to have any effect on the LVS.

EngGhaith, 2025-04-23:
Update:
the problem is in the netlist simplify.
I have tried to unconnect the sub! to either both capsor one of them and LVS works (it says that the caps aren't match to netlist, but no RuntimeError).

Also it is in my opinion related to sub! connection.....

KrzysztofHerman, 2025-04-28:
@EngGhaith 
could you please post both netlists.

KrzysztofHerman, 2026-04-27:
@EngGhaith  I have made a similar testcase 
```
.subckt scap c0 c1
*.PININFO c0:B c1:B
C1 c0 net1 sub! rfcmim w=7.0e-6 l=7.0e-6 wfeed=3.0e-6
C2 net1 c1 sub! rfcmim w=7.0e-6 l=7.0e-6 wfeed=3.0e-6
.ends
.GLOBAL sub!
```
[image]

[image]

**But still, although you do not have red warnings it fails on `   'Port mismatch \'$3\' vs. \'SUB!\'`**