IHP-GmbH/IHP-Open-PDK #460: Use of the PDK for photonics?

lucasgrjn, 2025-04-22:
Hi,

I am curious: is it possible to use this PDK to design some hybrid photonics/electronics chip?

Thanks for the response, and thanks for the PDK!

Lucas

KrzysztofHerman, 2025-04-25:
@lucasgrjn 
In general the possibility to integrate photonics elements in open source G2 technology is not possible. We are working on heterogenous integration, what will enhance the spectrum of possible integrations. 
You can also checkout other possibilities [here](https://www.ihp-microelectronics.com/services/research-and-prototyping-service/mpw-prototyping-service/sigec-bicmos-technologies)

lucasgrjn, 2025-04-25:
@KrzysztofHerman 
Thanks a lot for your response!

Unfortunately, since the SG25H5_EPIC process is not open-sourced, I don't have access to any data.
Do you have any plans to open it in a near future?

KrzysztofHerman, 2025-04-29:
@lucasgrjn 
Please subscribe and stay tuned at the fossi-char.org, we have a channel there: `ihp-sg13g2`.
At IHP there are ongoing discussions about open sourcing more technologies.