IHP-GmbH/IHP-Open-PDK #461: Question about maximum current in Rsil

EngGhaith, 2025-04-24:
Hi
In the docs there is no description, about how much current can Rsil drop.

[image]

It is only refers to 2.9, but 2.9 is for Rhigh. I think this reference is from an older version.....

It said limited by contacts, but in the layout there is always only one contact per pin, doesn't matter how long/wide is the resistor. Here is a screenshot from Klayout with w=5u/l=0.5u and w=0.5u/l=5u

[image]

Both resistors has only two contacts (one per pin)

Thank you in advance.
Ghaith.

KrzysztofHerman, 2025-04-25:
@EngGhaith 
Thank you for reporting. We will update the documentation. 
The value of max. current density is `1 mA/μm`

EngGhaith, 2025-04-25:
Thank you very much!
I think we could close this now ...

sergeiandreyev, 2025-04-25:
@EngGhaith, @KrzysztofHerman, I will prefer to have it opened until we update the documentation