IHP-GmbH/IHP-Open-PDK #462: SRAM macro models - A_DLY pin

phsauter, 2025-04-25:
The [SRAM documentation](https://github.com/IHP-GmbH/IHP-Open-PDK/blob/main/ihp-sg13g2/libs.ref/sg13g2_sram/doc/RM_IHPSG13_1P_64x64_c2_bm_bist.txt#L96) for the `A_DLY` pin simply states:
> Delay setting, adjustment of memory internal timings; recommended setting: Tie to 1

Its function does not seem to be further documented nor is it reflected in either the Verilog or Liberty models.  
Would it be possible to provide more information on its purpose and if it has externally observable differences (eg interface timings), also add it in the Liberty and/or Verilog models of the SRAMs?  

If this is not possible or would be too much effort I would recommend an assertion be placed in the SRAM Verilog model that catches it being tied to zero. 

We just caught this mistake in [Croc](https://github.com/pulp-platform/croc) and while the documentation clearly states it should be tied to 1, it is rather easy to miss at the moment.

phsauter, 2026-04-01:
> I hope tying the pin to 0 doesn't mean we'll get a brick - just taped out a design with A_DLY set to 0 because this is way too easy to miss and it should definitely be part of the simulation model if it really changes anything significant.

For us most of the macros still worked fine across two different tapeouts. It seems to depend on the SRAM macro itself and probably some manufacturing parameters.  
So it is still entirely possible that it works fine, lets hope!

sergeiandreyev, 2026-09-02:
Hi @phsauter, the proposed solution is merged in #1096 - could you please take a look and confirm so that we can close the issue?

phsauter, 2026-09-02:
Yes this matches what we do internally, thanks for the change!

closed 2026-09-02 by phsauter