IHP-GmbH/IHP-Open-PDK #463: Questions about HBT Layout

EngGhaith, 2025-04-26:
Hi!
I found this section in the documentation and have some questions about that:

[image]

Let us take this transistor as example:

[image]

1. Rule says that we are only allowed to use metals in the TRANS region. Are vias also allowed?
2. Are not we allowed to route in red hatched region even for metal 2? or only metal 1?
3. Which of these connections between B and C (Yellow/Pink/Both) are allowed (Both yellow and pink are on Metal 1 layer)?
4. Are these bipolar rules existed in KLayout-DRC or not? i.e. If DRC is clean, should the design be OK or not?

Thank you in advance.
Ghaith.

KrzysztofHerman, 2025-04-28:
@EngGhaith 
The purpose of the cited part of documentation is to prevent the users to modify the device especially the part related to the FEOL, what can affect the device resulting in a not working one. As for BEOL
1) Each metal shape will introduce parasitic effects. It will affect the performance but not kill the device itself. You can place vias 
2) Routing on metal1 and metal2 is allowed, consider EM coupling an parasitic effects. 
3) Both are allowed, just respect DRC's and keep in mind parasitic effects. 
4) Reviewing the [missing rules](https://github.com/IHP-GmbH/IHP-Open-PDK/blob/dev/ihp-sg13g2/libs.tech/klayout/tech/drc/MissingRules_maximal.md) there is only one related to filler `Afil.e` however the filler generation scripts seem to respect it.

EngGhaith, 2025-04-29:
Thank you very much!

closed 2025-04-29 by EngGhaith