IHP-GmbH/IHP-Open-PDK #469: question: why does LVS dont accept distributed rppd?

olisnr, 2025-05-03:
if i split an 7x2.5µm Rppd in two 3.5x2.5µm Rppd's, i get a LVS missmatch. with MOS-devices this wouldnt happen. so my question is: is this intended?

i tried with and without the LVS-option `Combine Devices`:

[image]

and got:

[image]

d-m-bailey, 2025-05-15:
@olisnr Looking at the code for combing resistors in `pdks/ihp-sg13g2/libs.tech/klayout/tech/lvs/rule_decks/custom_combiner.lvs`
```
class RESDeviceCombiner < RBA::GenericDeviceCombiner
  include DeviceCombinerMethods

  # Method to check and perform device combination
  def combine_devices(a, b)
    res_nets = [0, 1] # Using id instead of names

    # Check if same parameters

    # Check if terminals series or parallel (With same params)
    if supp_parallel(a, b, 0, 1)
      return false unless PARALLEL_RES
      return false unless %w[w l ps b].all? { |param| same_parameter(a, b, param) }

      a.set_parameter('m', a.parameter('m') + b.parameter('m'))
    elsif supp_series(a, b, 0, 1)
      return false unless SERIES_RES
      return false unless %w[w ps b m].all? { |param| same_parameter(a, b, param) }

      a.set_parameter('l', a.parameter('l') + b.parameter('l'))
    else
      return false
    end

    # Disconnect the second device and let the system clean it up
    res_nets.each { |term| b.disconnect_terminal(term) }

    true
  end
end
```
it appears that for parallel resistors with the same `w` and `l`, `m` values are summed.

If the final comparison is between `w * m` of the 2 devices, they should match, but if `w` and `m` are compared separately, they won't match. Unfortunately, I haven't been able to find the comparison logic.

olisnr, 2025-05-15:
@d-m-bailey yes, it seems that schematic and layout netlist are different handled for parallel resistors. i think it would be a better option to use **w** instead of **m** because if You draw a schematics with say `w=10u` and then make an layout with parallel `w=7u `and `w=3u` **m** wont work anyway...

olisnr, 2025-05-15:
are You also looking in parallel `cmim`?

d-m-bailey, 2025-05-15:
> are You also looking in parallel cmim?

@olisnr The parallel cmim reduction look ok as far as I can see. It doesn't appear to attempt to combine series devices.
```
class MIMCAPNDeviceCombiner < RBA::GenericDeviceCombiner
  include DeviceCombinerMethods

  # Method to check and perform device combination
  def combine_devices(a, b)
    # Check if both devices have the same net
    return false unless same_net(a, b, 'mim_top') && same_net(a, b, 'mim_btm')

    # Check if parameters are the same
    return false unless same_parameter(a, b, 'w') && same_parameter(a, b, 'l')

    # Combine by summing up 'm' parameter
    a.set_parameter('m', a.parameter('m') + b.parameter('m'))

    # Disconnect the second device and let the system clean it up
    b.disconnect_terminal('mim_top')
    b.disconnect_terminal('mim_btm')

    # Disconnect mim_sub terminal if present and model name includes 'rfcmim'
    b.disconnect_terminal('mim_sub') if b.name.downcase.include?('rfcmim')

    true
  end
end
```

d-m-bailey, 2025-05-15:
> @d-m-bailey yes, it seems that schematic and layout netlist are different handled for parallel resistors. i think it would be a better option to use w instead of m because if You draw a schematics with say w=10u and then make an layout with parallel w=7u and w=3u m wont work anyway...

I agree. I'm just not sure how to implement that in the current system. I was thinking that the initial input of the netlist could calculate the effective w = `w` * `m`, and then ignore m in the subsequent reductions and checks. It would be confusing though if the `m` parameter is listed and doesn't match. 

Incidentally, the current rule will not combine devices with `w` values that don't match.

olisnr, 2025-05-15:
@d-m-bailey i would think it should allow both C's have the same `w` and the same `l`, but also `w` and `l` could be exchanged. there is no `W/L` its `W*L` the order dosnt matter.

and the real problem in the LVS comes from the CDL-simplification. here there is no `m` used but the `w` and the `l` are both multiplied instead of only the `w`.

d-m-bailey, 2025-05-15:
@olisnr From what I can understand of the rules for `cap_cmim`, `w` and `l` are not compared so the order doesn't matter. `A`, `P`, and `m` are compared. Where `A` and `P` are calculated as `w * l` and `2 * (w + l)` respectively. 

So the current rules will not match equivalent capacitance areas. They will match only if `A`, `P`, and `m` are the same. 

This is probably not what is generally expected.

olisnr, 2025-05-16:
@d-m-bailey i dont understand the code of `def combine_devices(a, b)`, but compairing A as area and P as length of the perimeter is what i would expect. this seems perfekt for me.

then only the reference calculation seems wrong. can You look into the handling of multiple devices on the schematic side?

d-m-bailey, 2025-05-16:
@olisnr The following is my understanding.
```
  def combine_devices(a, b)
    # Check if both devices have the same net
    return false unless same_net(a, b, 'mim_top') && same_net(a, b, 'mim_btm')
```
Do not combine the devices if the connections are not the same (ie, parallel).
```
    # Check if parameters are the same
    return false unless same_parameter(a, b, 'w') && same_parameter(a, b, 'l')
```
Do not combine devices if `w` and/or `l` are different.
```
    # Combine by summing up 'm' parameter
    a.set_parameter('m', a.parameter('m') + b.parameter('m'))
```
If devices are parallel and have the same `w` and `l`, sum the occurences.
```
    # Disconnect the second device and let the system clean it up
    b.disconnect_terminal('mim_top')
    b.disconnect_terminal('mim_btm')
```
Remove the second device.

The reference (source) calculation is in `pdks/ihp-sg13g2/libs.tech/klayout/tech/lvs/rule_decks/custom_reader.lvs`
```
class DeviceCustomMIM < RBA::DeviceClassCapacitor
  def initialize(name)
    # clear terminals and parameters of resistor class
    clear_parameters
    clear_terminals
    clear_equivalent_terminal_ids
    
    # Adding params
    add_parameter(RBA::DeviceParameterDefinition.new('w', 'width', 0, false))
    add_parameter(RBA::DeviceParameterDefinition.new('l', 'length', 0, false))
    add_parameter(RBA::DeviceParameterDefinition.new('A', 'area', 0, true))
    add_parameter(RBA::DeviceParameterDefinition.new('P', 'perimeter', 0, true))
    add_parameter(RBA::DeviceParameterDefinition.new('m', 'multiplier', 1, true))

    # Adding terminals
    ter1 = add_terminal(RBA::DeviceTerminalDefinition.new("mim_top"))
    ter2 = add_terminal(RBA::DeviceTerminalDefinition.new("mim_btm"))
    ter1.name = "mim_top"
    ter2.name = "mim_btm"

    # Adding extra param & terminal for rfcmim
    return unless name.downcase.include?('rfcmim')

    add_parameter(RBA::DeviceParameterDefinition.new('wfeed', 'feed width', 0, true))
    sub_ter = add_terminal(RBA::DeviceTerminalDefinition.new("mim_sub"))
    sub_ter.name = "mim_sub"
  end 
end
```
My understanding is that `w` and `l` are input, but not compared (the final `false` parameter in `DeviceParameterDefinition` determines whether or not the parameter is compared.).
Parameters are mapped for each model type. For capacitors,
```
  def map_capacitor_params(device, model, params)
    device.set_parameter('w', (params['W'] || 0.0) * 1e6)
    device.set_parameter('l', (params['L'] || 0.0) * 1e6)
    device.set_parameter('m', params['M'] || params['MF'] || 1.0) if model.downcase.include?('cap_cmim')

    if model.downcase.include?('mim')
      device.set_parameter('A', (params['A'] || ((params['W'] || 0.0) * (params['L'] || 0.0))) * 1e12)
      device.set_parameter('P', (params['P'] || (((params['W'] || 0.0) + (params['L'] || 0.0)) * 2)) * 1e6)
    end
    return unless model.downcase.include?('rfcmim')

    device.set_parameter('wfeed', (params['WFEED'] || 0.0) * 1e6)
  end
```
So any time a device is created (extracted or read from source), the `A` and `P` parameters are calculated and `m` is initialized to 1 (for extracted devices) or the value in the source.

d-m-bailey, 2025-05-16:
As with resistors, I do not think that equivalent size checking is handled for `cap_cmim`.

olisnr, 2025-05-16:
if i see it correct, if the LVS script of the layout sees many parallel MOSFETs it calculates the new **w** it donsent make an **m** but for resistors it makes an **m**. 
the LVS on the schematic side does calculate **m** for MOSFETs and resistors and for the capacitors (even if its done wrong in this case (**l** is also calculated instead of let what it is)).

may be iHP should decide what the correct way is and then the layout and schematic side should do the same.

d-m-bailey, 2025-05-17:
> im wondering why the schematic and layout simplifier/combiner codes doesnt do the same. are there two different programs? or is the same program useded with different parameter?

@olisnr The program that reads the netlist needs to translate the device parameters into a standardized format while the netlist extractor will always extract individual devices. I think the netlist initialization routine is where the difference is introduced.

FaragElsayed2, 2025-08-28:
@omarmahxii Could you please check this?