IHP-GmbH/IHP-Open-PDK #471: Metal slits are not rightly recognized in the DRC/LVS

EngGhaith, 2025-05-04:
Hi
I have this simple test case, and it seems that LVS is not working correctly:
KLayout:

[image]

Xschem:

[image]

The LVS should throw an error, because the bottom plate of MIM on M5 should not be connected to the label on the left side of the slit.
However LVS is clean:

[image]

That means that the slit is not really excluded from M5 during LVS check.

Another example for DRC:

[image]

Slit is not recognized and therefore an error M3.b is thrown.

Files are here:

[TEST_SLIT.zip](https://github.com/user-attachments/files/20026902/TEST_SLIT.zip)

Thank you very much.
Ghaith.

akrinke, 2025-05-08:
Most Slit rules are supported by the DRC. The metal rules (such as `M3.b`) only look at the metal shapes and not the slit shapes. So you should fix the M3 violation underneath the slit shape.

EngGhaith, 2025-05-08:
@akrinke 
But Metal3.slit means, that there would not be Metal3 in the slit area.
Also there would not be any M3.b violation.

And LVS should throw error, because the slit would disconnect the bottom shhet of MIM from the net label/pin.

Also there is a strange behaviour here.

akrinke, 2025-05-08:
I understand the confusion. So we should subtract Metal3.slit before checking `M3.b`. @sergeiandreyev, @KrzysztofHerman What do you think?

sergeiandreyev, 2025-05-08:
Hi @akrinke, @EngGhaith, I'm checking on the internal PDK right now..

sergeiandreyev, 2025-05-08:
The behavior in commercial PDK/tools is the same (`M3.b` is flagged), so we're going to start an internal discussion on this

sergeiandreyev, 2025-05-08:
Yes, on the LVS testcase, I confirm, there should be Connectivity mismatch violation

sergeiandreyev, 2025-05-12:
> I understand the confusion. So we should subtract Metal3.slit before checking `M3.b`. [@sergeiandreyev](https://github.com/sergeiandreyev), [@KrzysztofHerman](https://github.com/KrzysztofHerman) What do you think?

Hi @akrinke, yes, we've got confirmation from technology perspective - there has to be no `M3.b` violation in this case - so the rules should be updated

EngGhaith, 2025-05-12:
@sergeiandreyev 
Could you please tell us the branch or PR, which we have to checkout, to get the new DRC/LVS?

EngGhaith, 2025-05-12:
@sergeiandreyev 

No problem.
You can open a PR for this specific topic, then we could subscribe that PR to know when you merge it to dev/main.

akrinke, 2025-06-10:
I have updated the DRC scripts in PR #538.

FaragElsayed2, 2025-08-28:
@omarmahXii Could you please take a look here?

FaragElsayed2, 2025-08-28:
Hi @EngGhaith 

I'm not sure what is the issue in LVS for that case?

The Metal.Slit is excluded from the drawing metal layers as shown here, there is no connections between parts.

But the LVS should be passed as expected.

[image]