IHP-GmbH/IHP-Open-PDK #472: Klayout : problems opening .gds files

giuliobrancali, 2025-05-05:
when opening .gds files( containing layoutdesign  made with the open pdk)  on klayout, sometimes klayout is not in editing mode and the result is that klayout delete all the file ...what could be the problem that generate this issue? Is there a specific setting to set?

VolkerMuehlhaus, 2025-05-06:
To start klayout in editor mode, you need commandline parameter -e

If you start klayout by double clicking on the GDS file, make sure the file association also starts klayout with -e, otherwise you are in viewer mode.

VolkerMuehlhaus, 2025-05-06:
> Thank you so much for the answer but is there a way to setup klayout to be always in editing mode?

In Klayout File > Setup, enable the "Use edit mode by default" checkbox