IHP-GmbH/IHP-Open-PDK #473: Poly resistor contact resistance modelling.

christoph-weiser, 2025-05-05:
# Poly resistor contact resistance modelling

## Issue

The poly resistor models include incorrect modelling for contact resistance.

## Affected Models

- Rsil
- Rhigh
- Rppd

## Issue Details

Here is an extract with relevant lines from the rppd model in "resistors\_mod.lib"

```
.subckt rppd 1 3
.param w=0.5e-6 l=0.5e-6 b=0 ps=0.18e-6 trise=0 m=1 sw_et=0
[..]
+rzspec=35e-6
+res_rzspec=2*rzspec/w
+rz=rzspec/w-(postsim>0)*rqrc/w
[..]

NR1 1 0 2 dt res_rppd L=leff W=weff m=m
+a1=a a2=a
+p1=p p2=p
+c1=1 c2=1
+trise=trise
+sw_et=sw_et

R2 2 3 R=res_rzspec TC1=-950e-6 m=m

.model res_rppd r3_cmc
[..]
+rc=rz
[..]
.ends rppd
```

It appears that the model is build from two components. 
A simple spice resistive element and the r3\_cmc compact model.

```
1   NR1   2    R2   3
x--vvvvv--x--vvvvv--x
     | 
     0
```

The contact resistance at one end of the resistor is modelled using the spice resistor R2. 

```
R2 2 3 R=res_rzspec TC1=-950e-6 m=1
```

the parameter res_rzspec that gives the resistance to the model is simply given by:
```
rzspec=35e-6
res_rzspec=2*rzspec/w
```

There is also modelling for contact resistance in NR1.
the parameter c1 (contacts at node n1 terminal) is set to 1.
the same is true for c2 (contacts at node n2 terminal). 
and rc (resistance per contact) is set to rz.
```
rz=rzspec/w-(postsim>0)*rqrc/w
```

There appears to be two seperate issues:

1.  The resistor is not symetrical so it is not irrelevant which
    way the resistor is placed when analyzing the impact of 
    parasitic capacitance to node 0.
2.  The contact resistance is accounting for both ends of the resistor hence the 
    multiplication by 2 in res_rzspec. But there is also contact resistance 
    modelled in the cmc model.

## Verification in Simulation

From the datasheet attached to the PDK we have the following information 

- RCRPPD = 35ohm\*um 
- RSRPPD = 260ohm

Hence a resistor with w=1e-6 and l=1e-6 should simply have a total 
resistance of:
```
R = 2*RCRPPD/w + 1*RCRPPD*(l/w) = 330 ohm
```
This assumes athat the parameter RCRPPD is required twice one for each side
of the resistor.

### Simulation netlist

``` 
.lib cornerRES.lib res_typ
.temp 27

vdd vdd 0 1
vss vss 0 0

XR1 vdd vss rppd w=1e-6 l=1e-6

.control
op
let i = abs(vdd#branch)
let r = vdd/i
print r
exit
.endc
.end
``` 

### Simulation result

with w=1e-6 and l=1e-6 we get: r=397.01

which is 67 ohm to high, roughly 2*RCRPPD.

dwarning, 2025-05-05:
I can confirm there is something wrong with the subckt model.

But I got slightly different (wrong) results with the shown test case above:
This for complete model
r = 4.009050e+02

Commented out the contact resistance R2 (only the r3 model is used)
r = 3.309359e+02

Commented out the poly res NR1 (only the contact res is between the nodes 1 3:
r = 7.000000e+01

Setting c1 and c2 to 0 (switch off r3 internal cont. res. calc.)
r = 3.309360e+02

So in the end we see that contact resistance is counted twice - by r3 model and R2 instance. That is the case for all poly res subckt models.

The models are transferred from other PDK lib 1:1 and seems never compared to spec.
I will provide an PR after further investigations.

christoph-weiser, 2025-05-05:
Thanks for checking it! 
Note that i just created a pull request with some changes regarding this issue. #476

sergeiandreyev, 2025-05-09:
Hi @christoph-weiser, this issue should be fixed in #481

closed 2025-05-09 by christoph-weiser