IHP-GmbH/IHP-Open-PDK #475: LVS and TLines

EngGhaith, 2025-05-05:
Hi
Is there any way to check TLines with LVS?
Or they should be shorted in the schematic before Netlist generation?

Thank you in advance.
Ghaith.

d-m-bailey, 2025-05-15:
@EngGhaith If you are using xschem, you can set `lvs_ignore=short` as a property on the devices (TLines, ammeters, etc.) that  you want the netlister to connect. You do not need to create a separate netlist.

I've fond that handling inductors in lvs can be tricky. I generally have to use some type of recognition layer or abstracted cell.

EngGhaith, 2025-05-15:
> [@EngGhaith](https://github.com/EngGhaith) If you are using xschem, you can set `lvs_ignore=short` as a property on the devices (TLines, ammeters, etc.) that you want the netlister to connect. You [...]

@d-m-bailey 
Could you explain more about inductors?
Do you have any example (Xschem and Layout)

d-m-bailey, 2025-05-15:
@EngGhaith Unfortunately, I haven't done anything with inductors in the iHP process yet. For the sky130 process, there were designs that had a layout cell that contained one inductor. The cell name was the name of the inductor model. The cell had ports that matched the inductor port names. I added a virtual metal resistor to allow the terminals to be extracted without shorting. Abstracting the cell effectively resulted in a inductor in the extracted netlist that could be matched to a similarly named inductor subcircuit in the source netlist.

d-m-bailey, 2025-07-02:
@EngGhaith If I understand correctly, you might be able to use metal resistors to logically separate the TLines from the connecting nets in LVS. For simulation, how are you modeling TLines?

Here's a layout where I added topmetal2 resistor to separate the connections to ground.

[image]

EngGhaith, 2025-07-02:
I am using the xyce ytransline model.
It is an RLC model of T-Line. I have designed a symbol for that.

But Topmetal2 resistor is nice idea.
I should then replace the T-Lines in schematic with resistors, or?

d-m-bailey, 2025-07-02:
@EngGhaith I don't think you need to replace the T-Line symbols. Just add an `lvs_ignore=short` property. If you add the `topmetal2` resistors, you can probably keep those in the simulation schematic because their resistance is negligible.

EngGhaith, 2025-07-03:
@sergeiandreyev 
@KrzysztofHerman 

And If I want to replace my T-Line within proper Metal/TopMetal resistor, then there is no symbol for that in xschem.
I would be very happy if you could produce such a symbol with following specs:

- Name of symbol: Rmet
- L as a parameter
- W as a parameter
- Layer as a parameter, which accept only following strings: [ 'M1', 'M2', 'M3', 'M4', 'M5', 'TM1', 'TM2']
- R value should be calculated automatically

If there is also a way to implement something like: `lvs_ignore=Rmet(L=1u,W=5u,Layer='TM1')`, then you can refer the designer to this solution to overcome a T-Line in the LVS.

This solution would be very satisfying for easy T-Lines (only a rectangular T-Line), because the designer can give the dimensions of that T-Line and the LVS would check the T-Line dimesions. If the designer want then to implement a complicated T.Line (e.g. with 90-degree bends, round bends or mitered bends), then he has split his T-Lines in schematics to multiple resistors and model each rectangular part as an Rmet. The bend could then be left as a "short" from LVS-view.

See this example of complex T-Line:

[image]

Such a T-Line with mitered bend could then modeled of a combination of 2 in series metal resistors: Rmet1 + Rmet2
Rmet1: L=70u, W=8u, Layer='TM2'
Rmet2: L=40u, W=8u, Layer='TM2'
The benefit here is that LVS-Check would check the length and width of my T-Line also.

You can go a further step of luxury if you make an gds to xschem extractor for T-Lines. Also I draw a T-Line in KLayout and give it maybe to a python script. This script would divide the polygon (it should be only one merged polygon) to rectangular sectors and each sector would be then modeled as an Rmet. A net list of several Rmet in series would be generated. If you introduce T-Line-Symbol (you can also modify the one from the attachment above), then the output of that script is a **one single xschem symbol** which has integrated netlist of the drawn T-Line.

See this tutorial for integrating netlist in a symbol:
https://xschem.sourceforge.io/stefan/xschem_man/tutorial_use_existing_subckt.html

I am concretly speaking about option 3:

[image]

Because you would have then **ONLY** one file, which is the generated xschem symbol.