IHP-GmbH/IHP-Open-PDK #484: LVS sees wrong placed net-label but thinks it dosent matter...

olisnr, 2025-05-10:
i have a miss placed label:

[image]

and LVS sees it, but shows all in the greenest green possible:

[image]

the files:

[LVS_WTF1.zip](https://github.com/user-attachments/files/20138855/LVS_WTF1.zip)

i use KLayout o.29.12 and the PDKs DEV branch

d-m-bailey, 2025-05-15:
@olisnr Can you verify that there is only one `in_p` text in the layout?

olisnr, 2025-05-15:
@d-m-bailey Yes. 
there is one off `in_p` `in_n` `out_p` `out_n` `VDD` `VSS` and only `in_p` is wrong placed. the rest of the circuit is ok

d-m-bailey, 2025-05-15:
@klayoutmatthias does klayout snap text to layers?

klayoutmatthias, 2025-05-16:
Hi @d-m-bailey,

No KLayout should not snap. But maybe the label is attached to some derived layer that looks different? Or there is a second label somewhere else?

Matthias

olisnr, 2025-05-16:
its a thing from the PDK there are metal-layers for drawing and others for filling, labeling and so on. the idea is: if there is some shape on a *.drawing layer and a text on the corresponding *.text layer inside this shape, the shape gets the net-label. 

this labels are used to tell the LVS what name a net or pin has.

d-m-bailey, 2025-05-16:
@klayoutmatthias thanks for commenting.

@olisnr can you check the metal1/pin layer? It may be that the text is over the pin layer and that the pin layer touches the drawing layer. This was a problem in sky130, so they added an extra check to flag pins without drawing layers. Maybe a similar DRC rule could be added to the klayout deck.

olisnr, 2025-05-16:
@d-m-bailey no, there is nothing on the Metal3.pin layer. the only layers of Metal3 that are used are: .via .drawing & .text

d-m-bailey, 2025-05-16:
@olisnr The layout displayed is `DiV_HBT_CMiM_100R.gds` but the LVS results are for `HBT_DiV_1024_14_lvs.gds`. Is `in_p` floating in `HBT_DiV_1024_14_lvs.gds` also?

olisnr, 2025-05-16:
@d-m-bailey yes.
i send You a corrected and a defect .GDS & the schematic .CDL so You can try it: 

[off_text.zip](https://github.com/user-attachments/files/20260135/off_text.zip)

d-m-bailey, 2025-05-17:
@olisnr I have verified that LVS passes even if the text is floating. Thanks for the sample data!

[image]

@klayoutmatthias using the data in the `off_text.zip `file above, I executed the 2 following commands using the ihp pdk lvs rule files.
```
python3 $PDK_ROOT/$PDK/libs.tech/klayout/tech/lvs/run_lvs.py --layout=HBT_DiV_1024_14.gds \
  --netlist=DiV_HBT_CMiM_100R.spice \
  --run_dir=ok \
  --topcell=DiV_HBT_CMiM_100R \
  --run_mode=deep \
  --spice_comments \
  --no_simplify \
  --combine_devices \
  --top_lvl_pins

python3 $PDK_ROOT/$PDK/libs.tech/klayout/tech/lvs/run_lvs.py --layout=HBT_DiV_1024_14_in_p__off.gds \
  --netlist=DiV_HBT_CMiM_100R.spice \
  --run_dir=ng \
  --topcell=DiV_HBT_CMiM_100R \
  --run_mode=deep \
  --spice_comments \
  --no_simplify \
  --combine_devices \
  --top_lvl_pins
```
The `top_lvl_pins` flag does this
```
netlist.make_top_level_pins if TOP_LVL_PINS
schematic.make_top_level_pins if TOP_LVL_PINS
```
Which turns all top level text into pins, right?

Here's the difference between the extracted layouts. (Note that in the sample data given, `in_n` is floating, not `in_p`).
```
% diff ok/HBT_DiV_1024_14_extracted.cir ng/HBT_DiV_1024_14_in_p__off_extracted.cir 
7d6
< * pin in_n
10c9
< .SUBCKT DiV_HBT_CMiM_100R VSS in_p VDD in_n out_n out_p
---
> .SUBCKT DiV_HBT_CMiM_100R VSS in_p VDD out_n out_p
33c32
< Q$11 \$22 in_n \$26 VSS npn13G2 AE=0.063p PE=1.94u AB=6.6795p PB=11.92u
---
> Q$11 \$22 \$16 \$26 VSS npn13G2 AE=0.063p PE=1.94u AB=6.6795p PB=11.92u
48c47
< Q$16 \$10 in_n \$25 VSS npn13G2 AE=0.063p PE=1.94u AB=6.6795p PB=11.92u
---
> Q$16 \$10 \$16 \$25 VSS npn13G2 AE=0.063p PE=1.94u AB=6.6795p PB=11.92u
```

So 2 questions.
1. During extraction, is there a way to flag text that does not overlap a connectivity layer?
2. Is there a way to flag missing ports in LVS?

klayoutmatthias, 2025-05-18:
I think I see what's the point.

I guess that the problem is that stray texts generate floating nets, but usually they are ignored. But assume they are turned into top level pins - which is an actual use case, when you have cell abstracts. Those are cells with ports only, but no internals.

So basically it's kind of hard to differentiate whether that is an error or not.

Regarding your questions:

1. Not in LVS, but in DRC

A boolean NOT between labels and polygons gives you the labels that are not on a polygon:

```
report("Labels without polygons")

l1_labels = labels(1, 0)
l1_geo = polygons(1, 0)
l2_geo = polygons(2, 0)

(l1_labels - l1_geo).output("L1 labels not on polygons on L1")
(l1_labels - l2_geo).output("L1 labels not on polygons on L2")
```

2. Such a check is conceptually difficult for a number of reasons. First, not all netlists define top level pins (e.g. SPICE without a top subcircuit). Second, the definition of a top level pin in layout is quite agnostic. Currently, LVS does not have the concept of a pin shape built in. Everything that is labelled on top level becomes a pin. This includes labels for internal nets.

Eventually, "top level pins" are not really hard facts and consequently they are not checked. In general, pins are not designed in. They rather come into existence, once you make a connection to a cell. For subcells they are checked of course, because you may have made a connection that is not intended. But for top level cells, there is no such fail mode and correspondingly top level pins are not considered in the compare step.

So bottom line is: to implement top level pin checking, first some pin concept has to be introduced. We may need such a concept in the course of PEX anyway, so there is some chance this will be done.

Matthias

d-m-bailey, 2025-05-18:
@klayoutmatthias thanks so much for the detailed analysis. I can see there is currently no solution that would cover all use cases.

I believe Calibre LVS has a directive - something like `LVS CHECK PORTS YES` - that enables a strict checking of ports. I don't know if something similar would be easy to implement in Klayout or not.

I tried running LVS with the ports switched and it passed (which wasn't expected).

[image]

How easy do you think it would it be for a end user to post process the `lvsdb` file to check for mismatches? 
Looks like the `P` lines are pins and the offsets in the matching `Z` section are relative to the respective `J` and `H` sections. So maybe the user could specify a top level and have a script that checks that the matching ports are actually the same.

klayoutmatthias, 2025-05-18:
Hi @d-m-bailey,

The idea of KLayout is to provide the bits and pieces and a programming language to combine them into the solution you want to have.

By default, LVS is net name agnostic - it will use net names as hints for evaluation shortcuts, but does not require net names to match or the nets to be labelled. In my opinion that is an advantage because it allows LVS in cases where there are not net names or net name matching is not given.

But it does not mean you cannot require equivalence of identically named nets in LVS. The key is the `same_net!` function (with an exclamation mark). Give a circuit and a net name, LVS will check if the net with the given name from layout and schematic match exactly.

So coming back to the "bits and pieces" idea, you can code the "require top level pins to be present both in layout and schematic in the same way" using a few lines of code:

```
top_circuit = schematic.top_circuit
top_circuit.each_net do |n|
  if n.pin_count > 0
    same_nets!(top_circuit.name, n.name)
  end
end
```

When there is no label in the layout, there is no error, if a matching net can be found attached to the pin. LVS will only know that something is connected to the pin with the right topology, so electrically, things are fine. But if you have no labels in the layout, it cannot tell whether you have mislabeled something.

Matthias

d-m-bailey, 2025-05-18:
@klayoutmatthias Amazing explanation. Totally makes sense.

We'll suggest that the iHP pdk includes your solution.

d-m-bailey, 2025-05-19:
@olisnr This may be something that we add to the iHP verification flow as it may not be easily implemented in Klayout.

Here's a short script, `port_check`, that I made to parse the `lvsdb` database to show ports that don't match.
```
#! /usr/bin/env bash

# Script to check port name matches in klayout lvsdb file

# Usage: port_check <lvsdb-file>

# Output: Mismatched ports in the format: layout_port source_port

# 1. Change unknown ports "()" in the lvsdb to "?".
# 2,3. Add space after "(" and before ")" to allow field parsing.
sed -e 's/()/?/g' -e 's/(/( /g' -e 's/)/ )/g' $1 |
        awk '
BEGIN {
        delete source;
        delete layout;
}
/^J/ { # Process layout
        while( ! /^ )/ ) { # until end of definition
                if ( /^ *P/ ) {
                        # Save port name at port order
                        layout[length(layout)] = $4;
                }
                getline;
        }
        layout["?"] = "?"; # Add entry for unknown ports
}
/^H/ { # Process source
        while( ! /^ )/ ) { # until end of definition
                if ( /^ *P/ ) {
                        # Save port name at port order
                        source[length(source)] = $4;
                }
                getline;
        }
        source["?"] = "?"; # Add entry for unknown ports
}
/^ *P/ && toupper(layout[$2]) != toupper(source[$3]) { # Case insensitive port comparison.
        # Ports are cross-referenced by respective port order.
        print layout[$2], source[$3];
}
'
```
And here are the results. Note that Klayout LVS passes for all these cases.
For the case where `in_n` was correctly placed.
```
% ./port_check ok/HBT_DiV_1024_14.lvsdb          
out_n OUT_P
out_p OUT_N
```
For the case where `in_n` was not over metal.
```
% ./port_check ng/HBT_DiV_1024_14_in_p__off.lvsdb
? IN_N
out_n OUT_P
out_p OUT_N
```
For the case where `in_n` and `in_p` were switched.
```
% ./port_check switch/HBT_DiV_1024_14.lvsdb     
in_n IN_P
in_p IN_N
out_n OUT_P
out_p OUT_N
```

klayoutmatthias, 2025-05-19:
I strongly discourage parsing the file. The format is internal and may change in the future.

klayoutmatthias, 2025-05-19:
In the above example, you mean that "DiV_HBT_CMiM_100R" is a subcicuit, not the top level circuit?

Port checking in subcircuits is not really important - if you have misspelled a port label, but connected it properly, the topology is correct. That is all that matters. Port checking makes sense on top circuits.

If "DiV_HBT_CMiM_100R" is the top circuit, why hard coding it? You have the name.

Regarding the case sensitivity - if you have SPICE input, net names should not be case sensitive. 

Where can I find the test case?

Matthias

olisnr, 2025-05-19:
@klayoutmatthias the test case:

[LVS_WTF1.zip](https://github.com/user-attachments/files/20280896/LVS_WTF1.zip)

You are right if the connections are correct, then the circuit is ok. 

but labels are a hint for me. if they aren't correct, they will cause only problems. i do some cells, and check it via LVS. in the end i but it together and check it also. if the labels arent correct then, i will rout a mess.

klayoutmatthias, 2025-05-19:
When I say "top level" I don't mean "chip". I mean the cell you are currently checking.

So when you work properly bottom up and do a check on every cell level, you will notice incorrect port labels. This is a recommended practice anyway.

Checking all ports in a design for proper labelling will not be helpful. In some cases you can even switch ports without bad side effects, like the A and B port of a CMOS NAND gate. Handling this case is built into LVS ("equivalent pins"), but violates the port naming rule.

Matthias

olisnr, 2025-05-19:
@klayoutmatthias yes thats all true. but i would like to find the wrong placed label when checking the current cell, and not when i but the things together. 

im new with LVS, and i think that little errors can produce a big  amount of error messages, that let me perplexed where to start. i would really like it to get the labels of cells correct at time of checking this cell only.

swapping input-pins or gates is also easier, if the cells are correct labeled. 

or do You see this different?

d-m-bailey, 2025-05-19:
> I strongly discourage parsing the file. The format is internal and may change in the future.

@klayoutmatthias Noted. But I think we need a way to verify that all the pins in the top level of the schematic are connected to corresponding labeled pads in the layout. In one sky130 design, there was a path that mistakenly did not connect to a port. The port was not connected to any device, and LVS passed because the netlists were topologically equivalent. On the top level of a design, the external pads should all be labeled to verify the coordinates of connectivity to outside world. With unlabeled pads, there is no way for LVS to verify the expected connectivity. Is this possible with the current version of Klayout?

> If "DiV_HBT_CMiM_100R" is the top circuit, why hard coding it? You have the name.
Yes, the top level layout name is probably available somewhere, but I wasn't sure how to extract it. In the interest of time, I just hard coded it here.

> Regarding the case sensitivity - if you have SPICE input, net names should not be case sensitive.
Are you saying that in Klayout, all these will give the same result?
``` 
      same_nets!("DiV_HBT_CMiM_100R", n.name, "DIV_HBT_CMIM_100R", n.name)
      same_nets!("DiV_HBT_CMiM_100R", n.name.downcase, "DIV_HBT_CMIM_100R", n.name)
      same_nets!("DiV_HBT_CMiM_100R", n.name.upcase, "DIV_HBT_CMIM_100R", n.name.downcase)
```
I know that case sensitivity is a factor in cell name comparison because this wouldn't work.
```
      same_nets!("DIV_HBT_CMIM_100R", n.name)
```

klayoutmatthias, 2025-05-19:
Case sensitivity seems to be an issue, that is correct. The extracted netlist is not case sensitive (as GDS texts have case). So you may have trouble matching an upper-case schematic net name to a lower case layout net name.

It is basically possible to turn off case-sensitivity in the extracted netlist

```
# Caution: will trigger extraction unless done already
netlist.case_sensitive = false
```

However, that has some side effects as names are normalized to upper case then. For example, device names need to be upper case too.

Another conceptual problem is maybe that "same_nets!" will not report an error if there is no layout net with that name. Topologically that still means the right circuitry is attached to the port, as this is checked by LVS. Also swapped labels will be detected, but not misspelled ones. So if your pins are called "A1" and "A2", but you label them "A" and "B", you will not get an error. With case sensitivity, you also won't see an error if schematic pins "A1" and "A2" are labelled "a2" and "a1". So case sensitivity is important here.

I will change the behaviour of "same_nets!" to become case insensitive if one of the netlists is. Also, I think it's okay to change the behavior of "same_nets!", so it would error if one of the nets is not found. That is just a problem for wildcards like `same_nets!("TOP", "*")` as schematic nets are always labelled and for internal nets there may not be a layout counterpart. Eventually you may not be able to use that function with wildcards in a meaningful way.

However, we're not out of options yet.

One generic solution that should allow you to add all kind of post-compare checks is based on the cross reference object. You can implement a very generic port check this way:

```
# here is the compare st
success = compare

if success

  # adds a port check

  messages = []

  xref = lvs_data.xref

  # find the circuit pair that forms the top level
  cpd = xref.each_circuit_pair.find { |p| p.first == netlist.top_circuit }
  
  # walk though the net pairs and check them
  xref.each_net_pair(cpd) do |np|
    schem_net, layout_net = np.first, np.second
    if schem_net.pin_count > 0 && schem_net && layout_net
      # check the names for nets where the schematic net is connected to a pin
      ln = layout_net.expanded_name.upcase
      sn = schem_net.expanded_name.upcase
      if sn != ln
        messages << "Layout net name differs from port name: #{sn} (schematic) vs. #{ln} (layout)"
      end
    end
  end

  # raise a compound message with all the mismatches
  if !messages.empty?
    raise "Port mismatches found:\n" + messages.join("\n")
  end    

end
```

This is essentially equivalent to parsing the lvsdb, but built into LVS and using the API.

Matthias

d-m-bailey, 2025-05-19:
@klayoutmatthias this may be exactly what we need! Really appreciate your time and precise in-depth explanations.

Quick unrelated question. The iHP rule set has these commands at the end.
```
# === COMPARISON ===
logger.info('Starting SG13G2 LVS Comparison')
compare

#================================================
#------------- COMPARISON RESULTS ---------------
#================================================

if !compare
  logger.info('xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx')
  logger.error("ERROR : Netlists don't match")
  logger.info('xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx')
else
  logger.info('==========================================')
  logger.info('INFO : Congratulations! Netlists match.')
  logger.info('==========================================')
end
```
Is this running `compare` twice?

d-m-bailey, 2025-05-20:
@klayoutmatthias the code you suggested almost worked perfectly. I made 2 changes.

1) I'm testing with version `0.29.4` and there doesn't appear to be a `top_circuit` associated with `netlist`. I instead used the name of the schematic.
```
  top_circuit = schematic.circuits_by_name($topcell)[0]
  cpd = xref.each_circuit_pair.find { |p| p.second == top_circuit }
```

2) I think the layout and schematic are reversed here, aren't they? This didn't flag missing layout ports.
```
    schem_net, layout_net = np.first, np.second
```
When I switched them, missing layout ports were flagged.
```
    layout_net, schem_net = np.first, np.second
```

klayoutmatthias, 2025-05-20:
And yes, the compare is run twice. One of the calls should be removed. It's just a waste of CPU time.

Matthias

klayoutmatthias, 2025-05-20:
I have created a ticket to add the above code as a new function for convenience, if you think it does the job.

d-m-bailey, 2025-05-20:
> I have created a ticket to add the above code as a new function for convenience, if you think it does the job.

Much appreciated!

atorkmabrains, 2026-01-15:
@sergeiandreyev and @KrzysztofHerman Please confirm if you need anything else on this issue.