IHP-GmbH/IHP-Open-PDK #489: Sealring generator script fails with Klayout

ismael-quside, 2025-05-13:
When I try to generate a sealring with klayout using this [script](https://github.com/IHP-GmbH/IHP-Open-PDK/blob/main/ihp-sg13g2/libs.tech/klayout/tech/scripts/sealring.py) I get the following error: 

```
ERROR: AttributeError: 'NoneType' object has no attribute 'layout'
  sealring.py:34
  sealring.py:70
```

The command to invoke the script is the following: `klayout -n sg13g2 -zz -r sealring.py -rd width=1300.0 -rd height=1300.0 -rd output=macros/sealring.gds.gz`

I tried several versions of klayout from the last year without success.

olisnr, 2025-05-14:
try the script from the DEV branch of the PDK:

use `git clone -b dev --recursive https://github.com/IHP-GmbH/IHP-Open-PDK.git`

ismael-quside, 2025-05-14:
I got the same error with `dev` branch. I can not see any changes in the script from `main` branch.
I tested dev branch with klayout 0.30.1 and 0.29.6

olisnr, 2025-05-14:
did You also try the sealring PCELL in KLayout without script? im using this:

[image]

select SG13 library in the bottom-left, and then drag the sealring into the layout-window

sergeiandreyev, 2025-05-14:
Hi @ismael-quside, please check the versions (commits) of the IHP OpenPDK repository and respective submodules:
```
$ git rev-parse HEAD
35dabf4b9ffe6b4697554e22b5416a0811e8daa4
...
$ git submodule status
 9a6e7c9d404f9f4334513a24788aa8454918d400 ../../../digital (heads/master)
 4c463c43991fb0967a824906518b66047bfb33c4 ../../python/pycell4klayout-api (heads/main-1-g4c463c4)
 cf1ff9bad0fb5338cf1c5b990b2b816b1ea01a64 ../../python/pypreprocessor (heads/master)
```
I cannot reproduce your issue with the latest `dev`:
```
andreev@IHP-OPDK:~/opensource/IHP-Open-PDK/ihp-sg13g2/libs.tech/klayout/tech/scripts$ klayout -n sg13g2 -zz -r sealring.py -rd width=1300.0 -rd height=1300.0 -rd output=macros/sealring.gds.gz
Warning: Box.destroy: already destroyed!
Warning: Box.destroy: already destroyed!
Warning: Box.destroy: already destroyed!
Warning: Box.destroy: already destroyed!
Warning: Box.destroy: already destroyed!
Warning: Box.destroy: already destroyed!
Warning: Box.destroy: already destroyed!
Warning: Box.destroy: already destroyed!
Warning: Box.destroy: already destroyed!
Warning: Box.destroy: already destroyed!
Warning: Box.destroy: already destroyed!
Warning: Box.destroy: already destroyed!
Warning: Box.destroy: already destroyed!
Warning: Box.destroy: already destroyed!
Warning: Box.destroy: already destroyed!
Warning: Box.destroy: already destroyed!
Warning: Box.destroy: already destroyed!
Warning: Box.destroy: already destroyed!
```

[image]

KLayout 0.29.12
Python 3.10.12

sergeiandreyev, 2025-05-14:
also, `KLAYOUT_HOME` and `KLAYOUT_PATH` env variables should be set properly in order to find the `SG13_dev` library:
https://ihp-open-pdk-docs.readthedocs.io/en/latest/install/installation.html#general-configuration

sergeiandreyev, 2025-05-14:
if I'm unsetting the `KLAYOUT_PATH` I'm getting the same error as in the issue description:
```
$ klayout -n sg13g2 -zz -r sealring.py -rd width=1300.0 -rd height=1300.0 -rd output=macros/sealring.gds.gz
ERROR: AttributeError: 'NoneType' object has no attribute 'layout'
  sealring.py:34
  sealring.py:70
andreev@IHP-OPDK:~/opensource/IHP-Open-PDK/ihp-sg13g2/libs.tech/klayout/tech/scripts$ echo $KLAYOUT_PATH

```

ismael-quside, 2025-05-14:
You are right @sergeiandreyev . I realized a bit late I wasn't configuring the environment properly. After that the script worked.

Thank you for your support and good addition for the script BTW.

closed 2025-05-14 by ismael-quside