IHP-GmbH/IHP-Open-PDK #493: DRC of small CMIMs

EngGhaith, 2025-05-14:
Hi!
I have a small MIM with: w=l=2u3.

I have used the PCell and had the following DRC-violation:

[image]

Should I add a wider box on TM1, or not because it is rightly generated?
If I made the upper plate on TM1 bigger to satisfy the rule, would I affect the capacitance a lot?

Thanks in advance.
Ghaith.

sergeiandreyev, 2025-05-14:
Hi @EngGhaith, we should have a limit set in the Pcell to prevent generating device layout w/ DRC violations, we will discuss this internally

sergeiandreyev, 2025-05-14:
well, these limits are not working (you can just try to draw a device w/ 1um size, it will crash w/ Python error) and the limit values are not defined correctly (if we can generate device layout that is not DRC clean, but satisfying the limits)

sergeiandreyev, 2026-08-28:
Hi @EngGhaith, regarding the capacitance: the dominant intentional capacitance is between the MIM top electrode and the `Metal5` bottom plate, across the thin MIM dielectric. The `Vmim` shapes are contacts connecting the MIM top electrode to `TopMetal1`; they do not define the active capacitor area.
Therefore, enlarging only the `TopMetal1` landing while leaving the MIM `w` and `l` unchanged should not change the intrinsic `cap_cmim`
model value. It can slightly change interconnect and fringing parasitics, which should be included by parasitic extraction when relevant.
Magic assignment of these parasitics was subsequently corrected in #803. For the reported 2.3 µm × 2.3 µm device, widening `TopMetal1` symmetrically to satisfy `TM1.a` is therefore a reasonable workaround.

sergeiandreyev, 2026-08-28:
Update after checking the current dev branch (`62607fba`, 2026-08-28):
Some related problems have been addressed. In particular, #640 fixed the rounding failure at the exact 1.14 µm minimum, and a newer parameter callback clamps GUI inputs below the configured minimum.
However, the original DRC problem reported here is still present. For a 2.3 µm CMIM, the PCell generates one 0.42 µm `Vmim` contact in
each direction. With the configured 0.42 µm `TopMetal1` enclosure on both sides, the resulting `TopMetal1` width is only:
`0.42 + 2 × 0.42 = 1.26 µm`
This is below the 1.64 µm required by `TM1.a`. The techFile value `cmim_minLW` also remains 1.14 µm, so satisfying the PCell limits still does not guarantee DRC-clean output.
Proposal (to be reviewed internally): fix the PCell to expand the `TopMetal1` landing to at least 1.64 µm in both directions, independently of the number of Vmim contacts. The generator should also validate undersized inputs directly rather than relying only on the GUI callback.

EngGhaith, 2026-08-29:
Hi @sergeiandreyev 
Since the main capacitance is defined somehow as the full sandwich (M5-MIM-TM1), then I would say that setting the minimum of TM1 in the PCell to 1.64u aligned to the center of the drawn MIM rectangular is the best solution.

sergeiandreyev, 2026-09-02:
Hi @EngGhaith, I've updated the `cmim` KLayout Pycell - could you please double check from your side so that we can close the issue?

sergeiandreyev, 2026-09-07:
@FrankVater, @adatsuk, @FarabiJamal, fyi