IHP-GmbH/IHP-Open-PDK #494: Problem with Netlist simplifier in KLayout

EngGhaith, 2025-05-14:
Hi!
I have this two in series connected capacitors:
### Xschem:

[image]

### Klayout:

[image]

### Xschem extracted LVS-Netlist:
```
.subckt C_3_4_MIMs c0 c1
*.PININFO c0:B c1:B
C1 c0 net1 cap_cmim w=7.0e-6 l=7.0e-6
C2 c1 net1 cap_cmim w=7.0e-6 l=7.0e-6
.ends
```
### Klayout extracted LVS-Netlist:
```
.SUBCKT C_3_4_MIMs c0 c1
C$1 c0 \$2 cap_cmim w=7u l=7u A=49p P=28u m=1
C$2 c1 \$2 cap_cmim w=7u l=7u A=49p P=28u m=1
.ENDS C_3_4_MIMs
```
### KLayout LVS result:

[image]

You can see that the reference netlist is not clean:
$C=\frac{\epsilon A}{d}$
if A get smaller, then C would be smaller.
here we have w=3u5 and l=14u --> A = w\*l = 49u ==7u\*7u ==> only one of the Caps?!
By connecting two caps in series it should be C_eff = C\/2, also A_eff = 49u\/2 = 24u5
Also the siimplifier is wrong!

I think you may turn that simplifier off or correct it.

Thank you in advance.
Ghaith.

sergeiandreyev, 2025-07-16:
Hi @EngGhaith, hm, maybe @FaragElsayed2 could take a look here?..

atorkmabrains, 2025-08-27:
Hi @sergeiandreyev and @EngGhaith,

We can't reproduce this issue.