IHP-GmbH/IHP-Open-PDK #496: Simulation results are inconsistent across identical runs after May 11, 2025 PDK update

rupokdas147, 2025-05-15:
After updating the PDK on May 11, 2025, I’ve observed non-deterministic simulation behavior with circuits that previously gave consistent results. Running the same netlist multiple times produces different results for the first few runs, and only after 3–5 simulations does the output stabilize and become repeatable.

This was not an issue prior to the update, and the netlist has not been changed in any way.

Environment:
Simulator: **Ngspice 44.2**, **Qucs-S 25.1.2**
OS: Ubuntu 24.04, Windows 10
PDK Version: Updated **after May 11, 2025** from **dev branch**
Circuit type: Custom design using [e.g., npn13g2 HBT, passive S-parameters, etc.]

This inconsistency also occurred for complex circuits, but for simplicity, I am including the PDK's example.

https://github.com/user-attachments/assets/41b5540c-1fe7-49f5-9984-6225a86e1b0e
https://github.com/user-attachments/assets/3ca5e717-caba-44cd-a6a6-6e63092a62a1

Look at the value of **i(Pr1)**.

dwarning, 2025-05-17:
Yes, that is our fault. Swapping default mm_ok to 1 in PR #453 let the repeated simulation runs act as mismatch simulations. This is the case for all elements with mismatch capability.
I am fearing we have to go back to the old selection approach with choosing different library entries for mismatch simulation, in case of hbt e.g. hbt_typ_mismatch.

The default setting should stay as ON with mm_ok=1.

Ikram-rs22, 2025-05-19:
@dwarning how to go back to the previous version or to turn off the mismatch variable? It is becoming impossible to simulate and design while mismatch parameter is ON.

dwarning, 2025-05-19:
Yes - we will switch back to the old concept asap.
But that simulations are not possible is definite wrong. The variations from run to run are much smaller then the corner and statistical variations. Your design should robust against that.

PR #505 (2025-05-21): Fix mismatch bug #496
Fixes #496

This PR fixes the problem of inconsistent results for repeated runs with identically circuit.

The drawback is that the user have to specify the dedicated library entry if he want execute mismatch simulations.
E.g. in case for hbt:
.LIB "$PDK_ROOT/$PDK/libs.tech/ngspice/models/cornerHBT.lib" hbt_typ_mismatch
or (dependent from .spinit)
.LIB cornerHBT.lib" hbt_typ_mismatch

The option to switch on/off mismatch simulation for selected devices in the circuit is preserved. Default setting is still "on".