IHP-GmbH/IHP-Open-PDK #498: sealring produces short-circuit in LVS

olisnr, 2025-05-15:
i have a strange problem:
a chip that seems ok without sealring in the LVS is short-circtuit in LVS if it has a sealring. but if tracing the nets in KLaout it looks ok with and without sealring. 

i was thinking that the difference of LVS and net-traceing is LVS looks also at the fill and tracing not, but if i delete all the fills it changes nothing.

but see Yourself 

[gds_and_spice.zip](https://github.com/user-attachments/files/20227533/gds_and_spice.zip)

there is still an LVS problem without sealring that comes from two parallel cmim's its not really a problem, its an error of the LVS handling of parallel cmim (at least i think this).

this is VSS without sealring

[image]

and this is VSS including sealring

[image]

d-m-bailey, 2025-05-19:
@olisnr could it be a short through psubstrate? The psubstrate connection is extracted in LVS, but is not normally part of the Klayout trace setup.

olisnr, 2025-05-20:
@d-m-bailey sure, but why does this need a sealring? and why does the net-tracing  dosent see's it?

i was searching for it, but didnt found anything. could You look at it?

d-m-bailey, 2025-05-21:
@olisnr If I change line `91` in `$PDK_ROOT/ihp-sg13g2/libs.tech/klayout/tech/lvs/rule_decks/general_derivations.lvs`
from 
```
ptap_holes = ptap.holes
```
to
```
ptap_holes = ptap.not(edgeseal_drw).holes
```
I get a clean LVS result (after connecting `REFADJ`, adding top text in the layout and top ports in the source, adding `VSS` to the `bondpad` instances in the source, and removing the bondpad instances for square instances in the source.)

olisnr, 2025-05-21:
@d-m-bailey thanks a lot! You fixed my chip!!!

You wrote that  "bondpad is extracted as a 2 terminal device" from the layout.? 

if i search in the 
[FMD_QNC_HBT_AMP_EH_extracted.cir.tar.gz](https://github.com/user-attachments/files/20374989/FMD_QNC_HBT_AMP_EH_extracted.cir.tar.gz)
i dont find any bondpad (that should have a `X` name i reckon).

where did You see a bondpad?

d-m-bailey, 2025-05-21:
@olisnr looks like the circuit you sent was the result of a flattened extraction. I extracted using the `deep` option.
Here's the subckt line of the file you sent.
```
.SUBCKT FMD_QNC_HBT_AMP_EH CGND|VSS C2|C3|VDD ien ld|op ihp VGND|vGND ip in ihn
+ iep REFADJ|vadj REF1.2|ip|vref sink in$1 C1|DRV_SHUNT|op SHUNT_GND|VSS_SHUNT
```

Looks like texts from all levels are present (and shorted) at the top level.

klayoutmatthias, 2025-05-21:
Hi @d-m-bailey,

regarding your question above: there are two bondpad cells, but only one schematic cell. For LVS there needs to be a 1:1 correspondence and the names need to match. As the names are "bondpad" and "bondpad$1", only the "bondpad" will match the schematic pad.

I don't see that the bondpad is extracted as a device. If it was, the difference in the cells would not matter - the shape would only be some parameter or hidden in an area value. In the test case and in LVS; the bond pad is simply a single-terminal subcircuit. As such, the name of the cell has to match. Otherwise, the subcircuit is flattened and simply vanishes.

I am worried about the "holes" thing. `ptap.holes` should also flag guard rings, which are pretty common. Is that considered? 

"holes" is an operation that usually has destructive side effects. It only applies in very rare situations such as detecting the inside areas of buried wells and only if they can't form donuts. But usually you have a mask layer for this, so there is no need to test the holes. I feel that inside LVS, holes are good for unexpected side effects only.

Matthias

olisnr, 2025-05-22:
modeling the a `bondpad` as a small `C` should work. what do You think?

d-m-bailey, 2025-05-22:
@klayoutmatthias 

> Also, I don't see that the bondpad is extracted as a device.

My bad. It's extracted as an empty subcircuit - which I consider to be a black-box "device".

> I feel that inside LVS, holes are good for unexpected side effects only.

ROFL. 
My patch eliminates the known problem of the enclosing `ptap` ring in the sealring, but as you stated, other `ptap` rings  might generate unexpected side effect.
@atorkmabrains

d-m-bailey, 2025-05-22:
@olisnr 
> modeling the a bondpad as a small C should work. what do You think?

I'm not sure I understand what you want to do. Are you suggesting that the schematic export the `bondpad` as a subckt that contains a capacitor? There will need to be a corresponding capacitor extracted from the layout.

If you don't care about the `bondpad`s, you can set `lvs_ignore=1` on the `bondpad` symbols in xschem. The current setup that netlist the `bondpad` as an undefined subckt should work with a little modification. 1) add a VSS terminal to the bondpad symbol in xschem so it netlists with 2 ports. 2) rename the square `bondpad` (maybe `bondpad2`) in both the layout and schematic. 3) figure out why Klayout is not extracting the square `bondpad` as a subckt.

There are several others ways to verify that there are actually connections to `bondpad` in the layout from the specified net. Unfortunately, I can't think of a way that does not require a rule change.

The magic tech file creates a pad tile from the passivation layer that requires text on a pad text layer. However, it does not support pins yet. I don't think the Klayout rules recognize pads or support pad text as a separate layer.

My opinion is that a specifically labelled pad with a substrate capacitor for top level external connections is a good idea. Without it, you run the risk of LVS passing but the pads not being connected to the intended circuit. 

Your layout currently has pads on the left that do not have text. The text is on the topmetal2 layer connected to the pads. Although this will pass LVS with the bondpad check, it does not check the location of the pads. Having the text over the pads (on a separate layer) will ensure that the pad at a specific location is actually connected to the expected signal. Without the `bondpad` subckt, LVS will pass even if the bondpads are not connected to these signals. See below.

[image]

olisnr, 2025-05-22:
@d-m-bailey the idea would be to have the bondpad as a capacitor to substrate. it would need that LVS extract the area of the pad and calculates the capacity and that the schematic bondpad symbol does the same. 
the schematic pad would have one pin and a connection to a global substrate net.

PR #511 (2025-06-12): Changed ptap_holes calculation
The holes function returns the inside of donut shapes. The seal ring is a ptap that surrounds the entire chip. When the sealring is included, the entire chip becomes `ptap_holes`. This change ignores the sealring ptap when calculating ptap_holes. For designs without the sealring, there is no change. 
Warning: Use of the holes function in LVS can cause unexpected side-effects when macros have enclosing tap rings.

Fixes #498

- [?] Tests pass <- This change is part of a larger set of changes that I'm working on for the TO_Apr2025 shuttle tests. The larger set of c [... cut]