IHP-GmbH/IHP-Open-PDK #509: Conflicts Between Verilog-A Component and PDK Elements Prevent Simulation

Amigom-MT, 2025-05-22:
Hi, 
I created a simple circuit composed of a variable voltage source, a component developed in Verilog-A and translated to OSDI using OpenVAF (in this case, a VCO), and an inverter built using transistors from the PDK "cornerMOSlv.lib mos_tt".

I tested both components separately, and they work perfectly.
When I tried to simulate them together, I initially got an error stating that "cornerMOSlv.lib" could not be found. I fixed this by providing the absolute path in the "Include Script" block.

I don’t understand why the block works without the absolute path when tested alone.
After fixing that issue, I’m now facing the following error:
"Ngspice started...

Warning: Nesting of subcircuits with parameters is only marginally supported!

Note: No compatibility mode selected!

Circuit: * qucs 25.1.1  /home/userproj/microelectronics/projetzigbee/zigbee/pll/pll_verilog_a/vco_fdiv_test.sch

Warning: Model issue on line:
  .model xsub1.xsg13_lv_nmos1.x1:sg13g2_lv_nmos_psp pspnqs103va type=+1 le ...
Unknown model type pspnqs103va - ignored

Warning: Model issue on line:
  .model xsub1.xsg13_lv_nmos1.x1:sg13g2_lv_pmos_psp pspnqs103va type=-1 le ...
Unknown model type pspnqs103va - ignored

Error on line:
  n.xsub1.xsg13_lv_nmos1.x1.nsg13_lv_nmos _net1 _net0 0 0 xsub1.xsg13_lv_nmos1.x1:sg13g2_lv_nmos_psp w=    1.500000000000000e-07     
l=    1.300000000000000e-07     nf=    1.000000000000000e+00     mult=    1.000000000000000e+00     as=    5.190000000000000e-14     
ad=    5.190000000000000e-14     ps=    9.920000000000000e-07     pd=    9.920000000000000e-07     dta=    0.000000000000000e+00     ngcon=2
could not find a valid modelname
    Simulation interrupted due to error!

Note: No ".plot", ".print", or ".fourier" lines; no simulations run"

Can you help me resolve or work around this conflict between the Verilog-A component and the PDK?

FYI : openvaf 23.5.0 - qucs 25.1.1 
[image]
complete circuit
[image]
Inverter
[image]
VCO

_Originally posted by @Amigom-MT in https://github.com/IHP-GmbH/IHP-Open-PDK/discussions/508_

dwarning, 2025-05-22:
My guess is that the psp103_nqs.osdi was not loaded. Perhaps your .spiceinit block is overwriting the pdk .spiceinit. Try to include both osdi files here.
If not success you must provide more information as the exported netlist and the va code of your VCO model.

Amigom-MT, 2025-05-23:
> My guess is that the psp103_nqs.osdi was not loaded. Perhaps your .spiceinit block is overwriting the pdk .spiceinit. Try to include both osdi files here. If not success you must provide more information as the exported netlist and the va code of your VCO model.

Ok, I added the absolute path to psp103_nqs.osdi in the .spiceinit block of the VCO, and I no longer get any errors related to that conflict.
Thanks!

closed 2025-05-23 by Amigom-MT