IHP-GmbH/IHP-Open-PDK #516: Klayout lvs with IHP : subckt ports not generated in .cir netlist

engrvip123, 2025-05-27:
Running Klayout-LVS using CLI with .cdl netlist from xschem for inverter circuit. 

`trg@opamp:~/IHP-Open-DesignLib/LNA_24GHz/design_data/lvs$ python3 run_lvs.py --layout=../inverter.gds  --netlist=../simulations/inverter.cdl  --run_dir=output_lvs1 --topcell=inverter  --top_lvl_pins`

Following `inverter_extracted.cir` netlist generated from klayout without subckt ports in netlist:

```
* Extracted by KLayout with SG13G2 LVS runset on : 27/05/2025 13:09

.SUBCKT inverter
M$1 \$5 \$2 \$3 \$I1 sg13_lv_pmos L=0.13u W=0.15u AS=0.1005p AD=0.1005p
+ PS=1.34u PD=1.34u
M$2 \$4 \$2 \$3 \$1 sg13_lv_nmos L=0.13u W=0.15u AS=0.1005p AD=0.1005p PS=1.34u
+ PD=1.34u
.ENDS inverter

```

Following CDL netlist generated from schematic:
```
* CDL netlist generated from SPICE

.SUBCKT inverter vdd inp out vss
* PININFO inp:B vdd:B vss:B out:B

M_M5 out inp vss vss sg13_lv_nmos W=0.15U L=0.13U M=1
M_M6 out inp vdd vdd sg13_lv_pmos W=0.15U L=0.13U M=1

.ENDS inverter
```

When importing `inverter.lvsdb` using Klayout's 'netlist browser' option, a cross-reference appears between two netlists. I have defined pins at layout terminals using `*.drawing` layers and assigned port names using `*.pin` layers. However, I am unsure why the `.cir` netlist generated by Klayout does not include subcircuit ports. which is leading to LVS failing ?

[image]

d-m-bailey, 2025-05-27:
@engrvip123 Can you share your gds? Does it have text and pins?

engrvip123, 2025-05-27:
@d-m-bailey Could not find any layer with datatype 65 in IHP layer [https://ihp-open-pdk-docs.readthedocs.io/en/latest/layout_rules/02_layer_table.html#layer-table](url)

Is it datatype 25 for text layer to be used?

d-m-bailey, 2025-05-28:
@olisnr there is a pin layer for gate poly (`5/2`), so a text layer would make sense.

@KrzysztofHerman @sergeiandreyev who makes the decisions on adding layer definitions to the pdk?