IHP-GmbH/IHP-Open-PDK #518: Running into Pin.f violations

dnltz, 2025-05-28:
OpenROAD-flow-scripts is able to generate macro blocks for Verilog modules. Those blocks have pin shapes for routing signals to in the parent design hierarchy.

[image]

When exporting a design to GDS, those pin shapes will be added to data layer 2 (`MetaX.pin`). According to the Layout Rules, pin areas ust be fully covered by drawing

```
Circuit designers should use only drawing purpose 0 (data types) for layouts. Only exception is pin purpose
2 for symbolic pins. Data type 2 (purpose pin) is used for symbolic connectivity information. Pin areas must
be fully covered by drawing. These rules are tested because pin areas are not used for mask generation
and a potential issue due to false postive LVS matchs.
```

Unfortunately, that is not possible with OpenROAD and designs with self-generated blocks will have `Pin.f` violations as a short part of the track is not covered by `MetalX.drawing`.

[image]

**Questions:**

1. Why is data layer 2 not used to generate masks and what's the purpose of data layer 2 then?
2. Can we export data layer 2 (``pin``) to data layer 0 (``drawing``)?

smunaut, 2025-06-10:
The `.map` file in ORFS is outdated. Replace it with the one from the PDK directly and this will fix this issue.

dnltz, 2025-06-17:
@smunaut after your explanation in the PR and running DRCs again, you're right. The `.map` was outdated.

closed 2025-06-17 by dnltz